ndokter/dsmr_parser #4: Issue with parsing telegram from Iskra ME382 meter

henkvdt, 2016-11-15:
I've tried to use the parser example with my ME382 meter, but something goes wrong in the parsing of the telegram. I've used the example code and change 4 to 2_2 for my meter.

The following error is logged on my screen:

> pi@raspberrypi:~ $ sudo python3 test.py

No line class found for: '/ISk5\2ME382-1003'
No line class found for: ''
No line class found for: '0-0:96.1.EC   13Y**'
No line class found for: '"'
No line class found for: 'kL(6K0:1.8.1(066EsCS[Wh)'
No line class found for: 'EkRsBs'
Traceback (most recent call last):
  File "test.py", line 11, in <module>
    for telegram in serial_reader.read():
  File "/usr/local/lib/python3.4/dist-packages/dsmr_parser/serial.py", line 71, in read
    yield self.telegram_parser.parse(telegram)
  File "/usr/local/lib/python3.4/dist-packages/dsmr_parser/parsers.py", line 67, in parse
    return super().parse(join_lines(line_values))
  File "/usr/local/lib/python3.4/dist-packages/dsmr_parser/parsers.py", line 32, in parse
    obis_reference, dsmr_object = self.parse_line(line_value.strip())
  File "/usr/local/lib/python3.4/dist-packages/dsmr_parser/parsers.py", line 47, in parse_line
    return obis_reference, parser.parse(line_value)
  File "/usr/local/lib/python3.4/dist-packages/dsmr_parser/parsers.py", line 135, in parse
    return CosemObject(self._parse(line))
  File "/usr/local/lib/python3.4/dist-packages/dsmr_parser/parsers.py", line 84, in _parse
    raise ParseError("Invalid '%s' line for '%s'", line, self)
dsmr_parser.exceptions.ParseError: ("Invalid '%s' line for '%s'", 'C\x02\x02\x02\x02\x02r\x02\x02\x12S[o\x1d\x7fjc\x0f1-0:2.8.2(00000.00ES[o\x1d\x7fj', <dsmr_parser.parsers.CosemParser object at 0x76868210>)

ndokter, 2016-11-15:
Thanks for creating the issue. I just want to let you know that ill look into it in the coming days.

aequitas, 2016-11-15:
I tested the EVEN parity with both ME 382  meters I have at my disposal and they both work. I think we can consider to make EVEN the standard parity for V2.2 as well.

ndokter, 2016-11-21:
I've made EVEN the default parity for v2.2 now. So now it's the default for both v2.2 and v4.

I'm not sure yet if we should define alternative settings yet and how they should be called (SERIAL_SETTINGS_V2_NONE or SERIAL_SETTINGS_V2_ALTERNATIVE?). If more people encounter issues with the current default settings, then it's worth considering defining multiple options that can be tried.

I will close the issue for now as the change has been applied. Note that a new version (0.4) has to be built, which i will do tomorrow probably.

closed 2016-11-21 by ndokter

aequitas, 2016-11-21:
@ndokter If I may give a hint, have a look at travis-ci, it automates testing/building for you public github projects for example: https://github.com/aequitas/python-rflink/blob/master/.travis.yml

ndokter, 2016-11-21:
Thanks, i will read up on it soon!

aequitas, 2016-11-21:
Btw, i already snuck in a travis file with one of the other pull requests ;) https://github.com/ndokter/dsmr_parser/blob/master/.travis.yml

All you need to do to enable testing for your repo is register with travis (using github account) and enable this repo in the settings.

ndokter, 2016-11-21:
I noticed :) Although i'm not familiar with how and when it builds yet.

aequitas, 2016-11-21:
It takes some time getting used to the concept, but Travis has good documentation to get you started and once you have one project setup it is trivial to get the rest running :)