ndokter/dsmr_parser #11: Improve library API

ndokter, 2016-12-26:
The current API can use some polishing to make it cleaner and more intuitive.

This issue focuses on improving the following points for now:

**telegram values**
After parsing a telegram in the form of a dict is returned. The values can be retrieved using defined constants.

It would probably be nicer to have a specific Telegram object that has some handy dynamic properties to retrieve the values with. The raw data will still be available.

**improve library API**
TODO. Not sure what changes are desired. General refactoring and simplifying library usage.

dennissiemensma, 2017-01-04:
Well, I can vouch for unittests/black box tests for this API.

This will allow you to easily detect non-backwards compatible changes in future releases, and highlight/warn people about that in future releases. 
Offcourse, this is dispite the fact that people **should** test any integration (or upgrade) of your library, before using it blindly anyway.

ndokter, 2019-12-21:
A true telegram object has been added: https://github.com/ndokter/dsmr_parser/pull/40

closed 2019-12-21 by ndokter