ndokter/dsmr_parser #21: Improve README

ndokter, 2017-01-21:
The README could describe the protocol client and async serial client. Also using the parsing logic without the clients (the libraries main goal) is not documented.

dajappie, 2017-02-02:
Hi Nigel, can you possibly also add an example how to parse a string-based telegram into a dictionary? Using Dennis' DSMR-reader I've split the datalogger-part using his API-example, so I can use the raw data also towards another system, but his implementation delivers a string with the telegram. Thanks!

ties, 2017-02-02:
I read my smartmeter over TCP. I do this using a small script that has two coroutines. The first connects to a socket and pushes the parsed telegram onto queue. The second coroutine pushes the data that I find relevant onto a MQTT broker.

This might be a helpful example for @dajappie and others.

It could use some cleanup. The current version of that script is available [in a gist](https://gist.github.com/ties/caa4d2b2ad35eaab65ec1d6f8c14fa0d)

ndokter, 2017-02-02:
In the tests are various examples, is that what you mean?
https://github.com/ndokter/dsmr_parser/blob/master/test/test_parse_v5.py

Snippet of it:
```
# It's a single string, just formatted differently for readability
telegram_str = (
    '/ISk5\2MT382-1004\r\n'
    '\r\n'
    '0-0:96.1.1(00000000000000)\r\n'
    '1-0:1.8.1(00001.001*kWh)\r\n'
    '1-0:1.8.2(00001.001*kWh)\r\n'
    '1-0:2.8.1(00001.001*kWh)\r\n'
    '1-0:2.8.2(00001.001*kWh)\r\n'
    '0-0:96.14.0(0001)\r\n'
    '1-0:1.7.0(0001.01*kW)\r\n'
    '1-0:2.7.0(0000.00*kW)\r\n'
    '0-0:17.0.0(0999.00*kW)\r\n'
    '0-0:96.3.10(1)\r\n'
    '0-0:96.13.1()\r\n'
    '0-0:96.13.0()\r\n'
    '0-1:24.1.0(3)\r\n'
    '0-1:96.1.0(000000000000)\r\n'
    '0-1:24.3.0(161107190000)(00)(60)(1)(0-1:24.2.1)(m3)\r\n'
    '(00001.001)\r\n'
    '0-1:24.4.0(1)\r\n'
    '!\r\n'
)
parser = TelegramParser(telegram_specifications.V2_2)
result = parser.parse(telegram_str)
```

Currently theres no auto version detection so youll have to supply a 'specification' dict for the DSMR version that tells the parser how to parse it.  

I've expiremented with detecting it (https://github.com/ndokter/dsmr_parser/blob/issue_22/dsmr_parser/parsers.py#L97), but haven't finished it yet.

dajappie, 2017-02-02:
Cool, that's it! Thanks! No real need for autodetection, standard DSMR 4 output so it'll be fine

closed 2017-02-27 by ndokter