ndokter/dsmr_parser #24: Home Assistant log shows exception

Phontana, 2017-05-29:
I see many lines in my home assistant (v. 0.45.1) log file, showing this:
```2017-05-29 19:41:35 ERROR (MainThread) [dsmr_parser.clients.protocol] failed to parse telegram
Traceback (most recent call last):
  File "/usr/local/lib/python3.6/site-packages/dsmr_parser/clients/protocol.py", line 100, in handle_telegram
    parsed_telegram = self.telegram_parser.parse(telegram)
  File "/usr/local/lib/python3.6/site-packages/dsmr_parser/parsers.py", line 47, in parse
    self.validate_checksum(telegram_data)
  File "/usr/local/lib/python3.6/site-packages/dsmr_parser/parsers.py", line 88, in validate_checksum
    expected_crc
dsmr_parser.exceptions.InvalidChecksumError: Invalid telegram. The CRC checksum '56532' does not match the expected '64335'
201```

I don't want the log to show these exceptions. How can I get rid of them?

syphernl, 2017-06-04:
I'm having similar issues with dsmr_parser 0.9 which was introduced in Home Assistant 0.46:
```bash
dsmr_parser.exceptions.ParseError: Telegram specification does not match telegram data
```

AlexMekkering, 2017-06-05:
In dsmr_parser 0.9, I have the exact same problem as @syphernl with protocol version V4 (Landis+Gyr ZCF110 / ZM F110 (DSMR 4.2)). dsmr_parser 0.8 does work correct.

AlexMekkering, 2017-06-05:
The problem at my side was that in my telegrams I don't have any lines matching L2 and L3 signatures, so at least the following signatures aren't present:
- VOLTAGE_SAG_L2_COUNT
- VOLTAGE_SAG_L3_COUNT
- VOLTAGE_SWELL_L2_COUNT
- VOLTAGE_SWELL_L3_COUNT
- INSTANTANEOUS_ACTIVE_POWER_L2_POSITIVE
- INSTANTANEOUS_ACTIVE_POWER_L3_POSITIVE
- INSTANTANEOUS_ACTIVE_POWER_L2_NEGATIVE
- INSTANTANEOUS_ACTIVE_POWER_L3_NEGATIVE

PR #25 fixes this for me

ndokter, 2017-06-05:
Do you perhaps also have an example telegram?

AlexMekkering, 2017-06-05:
Yes, I've zero'd out the identifiers and appended a fresh CRC16, but apart from that it was:
```
/XXX5XXXXXX0000000000

1-3:0.2.8(42)
0-0:1.0.0(170605083836S)
0-0:96.1.1(0000000000000000000000000000000000)
1-0:1.8.1(002327.805*kWh)
1-0:1.8.2(002000.399*kWh)
1-0:2.8.1(000000.000*kWh)
1-0:2.8.2(000000.000*kWh)
0-0:96.14.0(0001)
1-0:1.7.0(00.170*kW)
1-0:2.7.0(00.000*kW)
0-0:96.7.21(00001)
0-0:96.7.9(00000)
1-0:99.97.0(0)(0-0:96.7.19)
1-0:32.32.0(00000)
1-0:32.36.0(00000)
0-0:96.13.1()
0-0:96.13.0()
1-0:31.7.0(001*A)
1-0:21.7.0(00.170*kW)
1-0:22.7.0(00.000*kW)
0-1:24.1.0(003)
0-1:96.1.0(0000000000000000000000000000000000)
0-1:24.2.1(170605080000S)(01723.269*m3)
!FB3D
```

ndokter, 2017-06-05:
Although the build is currently 'failing' i made a new version (0.10). The log level of the checksum errors is changed to warning. Furthermore the pull request from @AlexMekkering has been merged which fixes an issue with the telegram signature being forced.

Please let me know if this works out

Phontana, 2017-06-08:
How can I test this? Do I have to wait for the next Home Assistant release?

ndokter, 2017-06-08:
You could manually install it with "pip install dsmr-parser==0.10"

Phontana, 2017-06-10:
I still get error messages with version 0.10:

```2017-06-10 14:54:50 ERROR (MainThread) [dsmr_parser.clients.protocol] failed to parse telegram
Traceback (most recent call last):
  File "/config/deps/dsmr_parser/clients/protocol.py", line 100, in handle_telegram
    parsed_telegram = self.telegram_parser.parse(telegram)
  File "/config/deps/dsmr_parser/parsers.py", line 47, in parse
    self.validate_checksum(telegram_data)
  File "/config/deps/dsmr_parser/parsers.py", line 88, in validate_checksum
    expected_crc
dsmr_parser.exceptions.InvalidChecksumError: Invalid telegram. The CRC checksum '46859' does not match the expected '4379'```

Phontana, 2017-06-10:
If the error is logged correctly on ERROR level, I can configure the logger to not show error messages of dsmr. With the current exception logging I don't have any option to hide these exceptions.

ndokter, 2017-06-10:
Im confused. Im can't test it easily myself, but the error should have been catched and logged as a warning. Could you give me the full file contents of "/config/deps/dsmr_parser/clients/protocol.py"?

Phontana, 2017-06-10:
This is the content of the file:

```"""Asyncio protocol implementation for handling telegrams."""

from functools import partial
import asyncio
import logging

from serial_asyncio import create_serial_connection

from dsmr_parser import telegram_specifications
from dsmr_parser.clients.telegram_buffer import TelegramBuffer
from dsmr_parser.exceptions import ParseError
from dsmr_parser.parsers import TelegramParser
from dsmr_parser.clients.settings import SERIAL_SETTINGS_V2_2, \
    SERIAL_SETTINGS_V4

def create_dsmr_protocol(dsmr_version, telegram_callback, loop=None):
    """Creates a DSMR asyncio protocol."""

    if dsmr_version == '2.2':
        specification = telegram_specifications.V2_2
        serial_settings = SERIAL_SETTINGS_V2_2
    elif dsmr_version == '4':
        specification = telegram_specifications.V4
        serial_settings = SERIAL_SETTINGS_V4
    else:
        raise NotImplementedError("No telegram parser found for version: %s",
                                  dsmr_version)

    protocol = partial(DSMRProtocol, loop, TelegramParser(specification),
                       telegram_callback=telegram_callback)

    return protocol, serial_settings

def create_dsmr_reader(port, dsmr_version, telegram_callback, loop=None):
    """Creates a DSMR asyncio protocol coroutine using serial port."""
    protocol, serial_settings = create_dsmr_protocol(
        dsmr_version, telegram_callback, loop=None)
    serial_settings['url'] = port

    conn = create_serial_connection(loop, protocol, **serial_settings)
    return conn

def create_tcp_dsmr_reader(host, port, dsmr_version,
                           telegram_callback, loop=None):
    """Creates a DSMR asyncio protocol coroutine using TCP connection."""
    protocol, _ = create_dsmr_protocol(
        dsmr_version, telegram_callback, loop=None)
    conn = loop.create_connection(protocol, host, port)
    return conn

class DSMRProtocol(asyncio.Protocol):
    """Assemble and handle incoming data into complete DSM telegrams."""

    transport = None
    telegram_callback = None

    def __init__(self, loop, telegram_parser, telegram_callback=None):
        """Initialize class."""
        self.loop = loop
        self.log = logging.getLogger(__name__)
        self.telegram_parser = telegram_parser
        # callback to call on complete telegram
        self.telegram_callback = telegram_callback
        # buffer to keep incomplete incoming data
        self.telegram_buffer = TelegramBuffer()
        # keep a lock until the connection is closed
        self._closed = asyncio.Event()

    def connection_made(self, transport):
        """Just logging for now."""
        self.transport = transport
        self.log.debug('connected')

    def data_received(self, data):
        """Add incoming data to buffer."""
        data = data.decode('ascii')
        self.log.debug('received data: %s', data)
        self.telegram_buffer.append(data)

        for telegram in self.telegram_buffer.get_all():
            self.handle_telegram(telegram)

    def connection_lost(self, exc):
        """Stop when connection is lost."""
        if exc:
            self.log.exception('disconnected due to exception')
        else:
            self.log.info('disconnected because of close/abort.')
        self._closed.set()

    def handle_telegram(self, telegram):
        """Send off parsed telegram to handling callback."""
        self.log.debug('got telegram: %s', telegram)

        try:
            parsed_telegram = self.telegram_parser.parse(telegram)
        except ParseError:
            self.log.exception("failed to parse telegram")
        else:
            self.telegram_callback(parsed_telegram)

    @asyncio.coroutine
    def wait_closed(self):
        """Wait until connection is closed."""
        yield from self._closed.wait()```

Phontana, 2017-06-10:
I compared the content of my file with the latest version and it seems that I have an older version of the file. I'm missing this part:

`except InvalidChecksumError as e:
            self.log.warning(str(e))`

However, when I ssh into my Docker container and run `pip install dsmr-parser==0.10` it says that it has already met that requirement.

Phontana, 2017-08-28:
This is fixed in 0.10. When will this be included in Home Assistant?

ndokter, 2017-08-29:
Sorry for not responding. Im glad that it's fixed in 0.10 I dont know when it will get included in Home Assistent as i'm not part of that project. Maybe you can make a pull request yourself?

Phontana, 2017-09-12:
Yes, I can do that, but I don't know for sure if all issues are fixed. Maybe we can make a 0.8.1 release with only the InvalidChecksumError log fix? We can then include that version in Home Assistant and there will be more time to fix the other issues users are experiencing in 0.9 and 0.10.

ndokter, 2017-09-18:
Is it also an option to test home assistent with the latest version? I cant do this myself currently though. But maybe you can manually upgrade the dsmr version with "pip install dsmr-parser==0.10" and see if your home assistent works correctly then?

Phontana, 2017-09-18:
Thank you for the suggestion. I tried that last weekend and it seems to work very well for me. Also check [this issue](https://github.com/home-assistant/home-assistant/issues/8128), since we're discussing the issue there as well.

PR #25 (2017-06-05): Support optional telegram signatures
This PR fixes #24 and adds support for optional signatures like the *_L2_* and *_L3_* ones (i.e. VOLTAGE_SAG_L2_COUNT). It just parses the ones that are present in the telegram data and doesn't care about missing ones.
closed 2018-09-23 by ndokter