ndokter/dsmr_parser #27: Creos Smart Meters P1 data encrypted

petzlux, 2017-09-18:
Hi,

Trying to find out if the Luxembourg smart meter which has a P1 port, can be integrated into Home Assistant.

In their [specifications](https://smarty.creos.net/wp-content/uploads/P1PortSpecification.pdf)  , they say the following:

> P1 protocol is a plain text protocol: it is required to secure the link to ensure end user data confidentiality. In deviation to DSMR P1, an encryption layer is added over the P1 protocol. This encryption layer is based on DLMS security suite 0 algorithm: AES128-GCM

Does your library support the decryption ?

ndokter, 2017-09-18:
Hey, there currently is no support for it built in sadly

grischard, 2017-11-06:
http://www.weigu.lu/microcontroller/smartyreader/ino/smartyreader.ino has arduino code that decodes the Creos smart meter data. http://www.weigu.lu/microcontroller/smartyreader/ is the full article. But like @petzlux I'd be happier to have this in python some day.

mweimerskirch, 2018-04-13:
I started working on this yesterday evening. My current idea is to write the decrypted data to a virtual serial port so existing software (like dsmr_parser or dsmr_reader) work without modification. Not sure how that will work out, but I guess I'll find out.

I currently have a proof of concept that prints out the decrypted DSMR telegrams on the command line: https://github.com/mweimerskirch/smarty_dsmr_proxy
This already works quite well. I'll try to find some time over the weekend to add comments to the code and continue working on it.

mweimerskirch, 2018-04-14:
Forwarding the decrypted data to dsmr_parser works now (via a virtual serial port created with "socat") and I can see everything show up correctly in Home Assistant. I added examples on how to do that to my repository. I also added more documentation to the code.
The next step would be to see if this could be directly integrated with dsmr_parser, but that's a subject for another day. In the meantime, I'd appreciate getting feedback from other Creos customers if the decryption works for them as well.

bossjl, 2018-09-20:
Hi, i try to parse the decrypted data directly and use it with the volkszaehler.org, but it seems that the parser does not read all the values from my (Lux - Creos) Smartmeter, specialy the most important 1.8.0 consumption value :) 
I did a print of the decrypted data, and the OBIS object array:

Framecounter
b'00016f0d'
/Lux5\XXXXXXXXX_D

1-3:0.2.8(42)
0-0:1.0.0(XXXXXXXXXXXXW)
0-0:42.0.0(5X41X73X30X33X37X03X31X23X30X93X)
1-0:1.8.0(000285.603*kWh)
1-0:2.8.0(000000.581*kWh)
1-0:3.8.0(000126.410*kvarh)
1-0:4.8.0(000000.528*kvarh)
1-0:1.7.0(00.176*kW)
1-0:2.7.0(00.000*kW)
1-0:3.7.0(00.008)
1-0:4.7.0(00.007)
0-0:17.0.0(77.376)
0-0:96.3.10(1)
0-0:96.7.21(00110)
1-0:32.32.0(00004)
1-0:52.32.0(00004)
1-0:72.32.0(00004)
1-0:32.36.0(00001)
1-0:52.36.0(00001)
1-0:72.36.0(00001)
0-0:96.13.0()
0-0:96.13.2()
0-0:96.13.3()
0-0:96.13.4()
0-0:96.13.5()
1-0:31.7.0(000*A)
1-0:51.7.0(000*A)
1-0:71.7.0(000*A)
!99DE

{'\\d-\\d:96\\.13\\.0.+?\\r\\n': <dsmr_parser.objects.CosemObject object at 0xb6293c10>, '\\d-\\d:0\\.2\\.8.+?\\r\\n': <dsmr_parser.objects.CosemObject object at 0xb67e0f90>, '\\d-\\d:2\\.7\\.0.+?\\r\\n': <dsmr_parser.objects.CosemObject object at 0xb67e0e70>, '\\d-\\d:72\\.32\\.0.+?\\r\\n': <dsmr_parser.objects.CosemObject object at 0xb67e0eb0>, '\\d-\\d:1\\.7\\.0.+?\\r\\n': <dsmr_parser.objects.CosemObject object at 0xb67e0f10>, '\\d-\\d:72\\.36\\.0.+?\\r\\n': <dsmr_parser.objects.CosemObject object at 0xb67e07b0>, '\\d-\\d:1\\.0\\.0.+?\\r\\n': <dsmr_parser.objects.CosemObject object at 0xb67e0f50>, '\\d-\\d:52\\.36\\.0.+?\\r\\n': <dsmr_parser.objects.CosemObject object at 0xb67e0e10>, '\\d-\\d:32\\.36\\.0.+?\\r\\n': <dsmr_parser.objects.CosemObject object at 0xb6163550>, '\\d-\\d:32\\.32\\.0.+?\\r\\n': <dsmr_parser.objects.CosemObject object at 0xb67e0e50>, '\\d-\\d:52\\.32\\.0.+?\\r\\n': <dsmr_parser.objects.CosemObject object at 0xb67e0ef0>}

I'l try to investigate this further....

ndokter, 2018-09-20:
@bossjl it seems that outside of the tariff values (1.8.1 and 1.8.2), theres also a 1.8.0 which indicates "Total imported energy register (P+)". These have indeed not been added as i wasn't aware of them. 

It should be added to https://github.com/ndokter/dsmr_parser/blob/master/dsmr_parser/obis_references.py and to the V5 specification in https://github.com/ndokter/dsmr_parser/blob/master/dsmr_parser/telegram_specifications.py but im currently not quite sure of the implications right now

bossjl, 2018-09-21:
Many thanks @ndokter 
It's working fine once I added the definitions to those files.
If those additions might impact depending projects, i could create a Lux specific fork

ndokter, 2018-09-21:
Because they are part of the v5.x specification, i think you should be added. Maybe you can make a pull request and i can incorporate them?

bossjl, 2018-09-21:
Done, i don't know if the name that to the obis reference is ok :) ... ELECTRICITY_IMPORTED_TOTAL

closed 2018-09-23 by ndokter

ndokter, 2018-09-23:
@bossjl thank you, ive build a new version with your changes. Havent tested them myself.

chrisnexxtlab, 2018-10-02:
I am glad, you all succeeded in reading your smart meter data. However, you might find this useful: https://www.nexxtlab.lu/smarty-dongle/

bossjl, 2019-06-18:
FYI: on the hardware side, the Creos counter feeds back 5V into the FTDI USB Serial Device converter , and feeding it back to the raspberry, i think this is not really healthy (The Pi hangs sometimes), so i soldered a diode onto the 5V wire of a cheap USB hub, and plugged it between the RPi and the FTDI converters.

koosoli, 2022-02-18:
Hi,
I am on HA 2022.2.8 on Rasbarry Pi 3. I live in Luxembourg and my meter is a Sagemcom T210D.
A bought this cable "DSD TECH SH-RJ12A USB to RJ12 6P6C Serial Cable with FTDI Chip Compatible for Iskra AM550 Sagemcom XS210" https://www.amazon.de/gp/product/B07R2L22GW/ref=ppx_yo_dt_b_asin_title_o00_s00?ie=UTF8&psc=1 that I connected to one of the USB ports.
The official core DSMR Slimme integration is unable to connect to it unless I change to v5 instead of v5L but then I can not put in my P1 encryption key that all Luxembourgish meters require to read the signals. So I get no values..
Can anybody help?
Thanks

jeasch01, 2023-09-20:
Hi,

@koosoli I have the same issue, have you found a solution in the meantime?
Any help is appreciated!

Thank you