ndokter/dsmr_parser #34: How to get the HOURLY_GAS_METER_READING timestamp value

baskapteijn, 2019-01-31:
It will probably be clear from my question that I'm no Python expert, but I'm struggling the to get the hourly gas timestamp value from the telegram parser. Is there a possibility to get this timestamp in my application?

This is a snippet of my current application, around which I would like to get the timestamp:

```
    for telegram in serial_reader.read():
        # Get the raw gas and electricity readings
        gas_reading = telegram[obis_references.HOURLY_GAS_METER_READING]
        elec_reading = telegram[obis_references.CURRENT_ELECTRICITY_USAGE]
```

ndokter, 2019-01-31:
Im not 100% sure, but can't you combine both HOURLY_GAS_METER_READING and P1_MESSAGE_TIMESTAMP to get the current gas reading value with measured time?

baskapteijn, 2019-01-31:
No the P1 message timestamp is de timestamp when the telegram has been send out on the uart. The gas reading part itself also contains a timestamp, and this is usually a timestamp that's only incremented each hour. See the spec:

https://www.netbeheernederland.nl/_upload/Files/Slimme_meter_15_a727fce1f1.pdf

page 22

Raw output on my console is, for example:

0-1:24.2.1(190130130000W)(04390.844*m3)

Where the first part within () is the gas reading timestamp.

But again, I don't know how to access it when using your dsmr_parser.

ndokter, 2019-01-31:
Yes, youre right about that

gas_reading = telegram[obis_references.HOURLY_GAS_METER_READING]

I think gas_reading is an 'MBusObject' which also has a 'datetime' value which corresponds to the value you mean

baskapteijn, 2019-01-31:
Ah I see, it appears that I can use `telegram[obis_references.HOURLY_GAS_METER_READING].datetime`.

Thanks for the info!

closed 2019-01-31 by baskapteijn