ndokter/dsmr_parser #35: UnicodeDecodeError: 'ascii' codec can't decode byte 0xff in position 0: ordinal not in range(128)

dartheide, 2019-02-22:
I use ser2net to send de P1 telegram to home-assistent, and it gives the following error.

File "/usr/local/lib/python3.6/site-packages/dsmr_parser/clients/protocol.py", line 83, in data_received
    data = data.decode('ascii')
UnicodeDecodeError: 'ascii' codec can't decode byte 0xff in position 0: ordinal not in range(128)

In protocol.py line 83 I changed the line from   data = data.decode('ascii') to data = data.decode('ascii', 'ignore'), and that solves the problem.

Gerben321, 2019-02-27:
Ah exactly what I am getting it seems: https://community.home-assistant.io/t/dsmr-help-needed/102063

I'll try this out when I get home today. Sounds good, thanks in advance.

Gerben321, 2019-03-01:
Just tried it, I no longer get the error, but It doesn't seem to work in Home Assistant. No data appearing.

dartheide, 2019-03-01:
For the data question, that more fore the home assistant GitHub page. Place your question there with a logfile.  ( Enable logging, https://www.home-assistant.io/components/logger/)

Gerben321, 2019-03-01:
I think it's more likely that your 'fix' only masks the problem, and doesn't really fix it. Nothing useful in my log.

dartheide, 2019-03-01:
It's fix the problem for ser2net. It's not a fix for other (configurations?) problems in home assistant.

Gerben321, 2019-03-01:
Can you show me your configuration? I've followed the HASS docs. Maybe also your ser2net config, so I'm sure it's not that. I assume the docs are right...

dartheide, 2019-03-01:
Placed my configuration in https://community.home-assistant.io/t/dsmr-help-needed/102063/2

ndokter, 2019-03-02:
Can you include the data from telnet? There might be some weird byte data that i havent taken into account

Gerben321, 2019-03-02:
Sure, I actually posted it on the Hass forum. https://community.home-assistant.io/t/dsmr-help-needed/102063/5

ndokter, 2019-03-02:
Could you include it as text here? Maybe i can try it out using a unit test

Gerben321, 2019-03-02:
Sure thing. I've copied a large part, so not sure where the start and end is!

`        ¯øíøµìçæ°°±°4438±¹±¹°
   ¡¶Á³´                                    °­±º24.2®±¨±¹°³°²±±°µ°µ×©¨°°°²¸®¸´³ªí³©¶®±®°¨´·³°³°³³³¹³°3031³¸³´³±³¸³°³µ³²³±³¸©                   °­±º²´®±®°¨°°³©
                                          ±­°º¶²®·®°¨°°®°°°ªë÷¹
                     ±­°º´²®·®°¨°°®°°°ªë×©
±­°º²²®·®°¨°°®°°°ªë×©                    ±­°º¶±®·®°¨°°®³¹´ºkW)
                    ±­°º´±®·®°¨°°®±°µªë÷©                                      ±­°º²±®·®°¨°°®°±·ªë÷©                                          ±­°º·±¾·.0(°°±ªÁ©
                                             ±-0ºµ±®·®°¨°°°ªÁ©
                            ±½0º31.7®°¨°°°ªÁ©
         ±­°º·²¾·¾°(237®´ªÖ©                                          ±­°ºµ²®·®°¨²³¶®µªö©                                          ±-°º³²®·®°¨²³·®²ªÖ©
                                      °­°º¹¶®±³®°¨©
                    ±­°º·²®³¶®°¨°°°°°©
  ±­°ºµ²®³¶®°¨°°°°°©                                            ±­°º³²®³¶®°¨°°°°°©                                            ±-0:7²®³²®°¨°°°°³©
                            ±­°ºµ²¾³²¾0(0°°°³©
          ±­°º³²®³²¾°¸000°³©      ±­°º¹¹®¹·®°¨±©¨°­°º¹¶¾·¾19)¸0°°±°±°±°°°°×©¨°°°°°°±¸±¹ªó©       °­°º¹¶®·®¹¨°°°°²©                                             0­°º¹¶®·®²±¨°°°°¶©                                           ±­°º²®·®°¨°°®°°°ªë×©
                                        ±­°º±®·®°¨°°®µ±¶ªë×©
                       °­°º¹¶®±´®°¨°°°±©                                      ±­°º²®¸®²¨°°°°°°®°°°ªë×è©                              ±½°º2.8®±¨°°°°°°®°°°ªë×è©
                            ±­°º±¾¸¾2¸000°±´®±²°ªë×è©
   ±­°º±®¸®±¨°°°°·²®³·´ªë÷è©         °­°º¹¶®±®±¨´µ³°³°³µ³²3030³´³³³¸³±³¹³±³¹³°³±³¸©          °½°:1.°®°¨±¹°³°²±±°¸±°×©
±­³º°®²®¸¨µ°©                                        ¯ØÍøµìçæ°°±°´´³¸±¹±90
                                                ¡°²²Á
         °­±º²´®²®±¨±¹°³°²±±°µ°µ÷©¨°°°²8.843ªí³©¶®±¾°¸´·3030³³³¹³°³°³±³¸³´³±³¸³°³µ³²³±³¸©                   °­±º²´®±®°¨°°³©
       ±­°º¶²¾·¾°¸00.0°°ªë×©                                      ±­°º´²®·®°¨°°®°°°ªë÷©                                      ±­°º²²®·®°¨°°®°°°ªë÷©
   ±­°º´±®·®°¨°°®±°µºû÷¹±­°º¶±®·®°¨°°®³¹±ªë÷©                 ±-0:²±®·®°¨°°®°±·ªë×©                                          ±­°º·±®·®°¨°°±ªÁ©
                            ±­°ºµ±®·®°¨°°°ªá¹
           ±­°º³±®·®°¨°°°ªÁ©                                            ±­°º·²®·®°¨²³·®³ªö©                                          ±½0ºµ²¾7.0¨²³¶®´ªÖ©
                                  ±­°º³²¾·¾0(2³·®±ªÖ©
                     °-0º¹¶®±³®°¨©
   ±­°º·²¾³¶.0(0°°°°©                                            ±½°:52.3¶®°¨°°°°°©                                            ±­°º³²®³¶®°¨°°°°°©
                             ±­°º·²®³²®°¨°°°°³©
           ±­°º5².32.°¨°°°°³©                                            ±­°º³²®³²®°¨°°°°³©      ±­°º¹¹®¹·®°¨±©¨°­°º¹¶®·®±¹¹¨°°°±°±°±°°°°÷©¨°°°°°°±¸±¹ºó¹
°½°º96.·®¹¨°°°°²©)                                          ±­°º²®·®°¨°°®°°°ºkW)
                                        ±­°º±®·®°¨°°®µ±³ªë÷©
                       °­°º¹¶¾±´¾°(000±©                                      ±­°º²®¸®²¨°°°°°°®°°°ªë÷è©                              ±­°º²®¸®±¨°°°°°°®°°°ªë÷è©
                            ±­°º±®¸®²¨°°°°±´®±²°ªë×è©
   ±­°º±®¸®±¨°°°°·²¾³74*ë×è©         °­°º¹¶.1.±¨´µ³°³°³µ³²³°³°³´³³³¸³±³¹³±³¹³°³±³¸¹          °-0º±®°®°¨±¹°³°²±±°¸°¹×©
±­³º°®²®¸¨µ°©                                 ¯ØÍøµìçæ°°±°´´³¸±¹±90
                                    °½±º9¡Ã°¹²¨´·³°³°³³³¹³°³°³±³¸³´³±³¸³°³µ³²³±³¸¹°½±:²´®²®±¨±¹°³°²±±°µ°µ÷©¨°°°²¸¾¸´³*m3)              ±­°º´²®·®°¨°°®°°°ºûW)
±­°º¶²®·®°¸°°¾000*ë×©             ±­°º²²®·®°¨°°®°°°ªë÷©
             ±½°º¶1.7.0¨°°®³¸¹ªë×©                                      ±­°º´±®·®°¨°°®±°´ªë÷©                                      ±­°º²±®·®°¨°°®°±·ªë÷©
                                  ±-0:7±®·®°¨°°±ªÁ©
                 ±­°ºµ±®·®°¨°°°ªá¹
±­°º³±®·®°¨°°°ªÁ©                                  ±­°º·²®·®°¨²³·¾²*V)
                                ±­°ºµ²®·¾°¸²³¶.4*Ö©
             ±­°º³²®·®°¨²³·®±ªö©
°­°º¹¶®±³®°¨©²®³¶®°¸°°°°0)                                            ±­°ºµ²¾36.0(°°°°°©                                            ±­°º³²®³¶®°¨°°°°°©
                                  ±­°º·²®³²®°¨°°°°³©
                ±­°ºµ²®³²®°¨°°°°³©                                            ±­°º³²®³²®°¨°°°°³©      ±­°º¹¹®¹·®°¨±©¨°­°º¹¶®·®±¹¹¨°°°±°±°±°°°°÷©¸°°°°0018±¹ªó©
     0-°º¹¶®·®¹¨°°°°²©                                             °­°º¹¶¾·¾²1(00°°¶©                                          ±­°º²®·®°¨°°®°°°ªë÷©
                           ±­°º±®·®°¨°°®µ±°ªë÷¹
          °-°º¹¶®±´®°¨°°°±©                                      ±­°º²®¸®²¨°°°°°°.00°ªë×è©                              ±­°º²®¸®±¨°°°°°°®°°°ªë÷è©
               ±­°º±®¸®²¨°°°°±´¾±20*ë×è©                              ±­°º±®¸®±¨°°°°·²®³·´ªë÷è©      ±­³°­°º¹¶®±®±¨´µ³°³°³µ³²³0303´³³³¸³±³¹³±³¹³°³±³¸©
°½°:1.0.°¨±¹°³°²±±°¸°¸×©
¿øíø5LGF0°±°´´³¸±¹±¹°     ±0EAÂ                                    °­±º²´®²®±¨±¹°³°²±1050µ×©¨°°°²¸®¸´³ªí³©¶®±®°¨´·³°³°³³³¹³°³°³±³¸³´³±³¸³°³µ3231³¸©
      °­±º²´®±®°¨°°³¹                                            ±­°º¶²®·®°¨°°¾°°0*k×©                                      ±­°º´²®·®°¨°°®°°°ªë÷©
                       ±­°º²²®·®°¨°°®°°°ªë÷©
`

ndokter, 2019-03-02:
Thank you, but I don't think i can do anything with this. Can you paste the data you get from CU instead?

Gerben321, 2019-03-02:
Here's the output. It has one full telegram, and a bit on top and at the bottom, since 3 didnd't fit on my screen. 
[cu.txt](https://github.com/ndokter/dsmr_parser/files/2921998/cu.txt)

ndokter, 2019-03-02:
I think what happens is that reading with CU, gets you the proper 'ascii' encoded text. But not when using the dsmr parser serial client.

There was a similar case here https://github.com/ndokter/dsmr_parser/issues/4 and the parser has multiple settings for the serial connection located here dsmr_parser/clients/settings.py are you able to try these different settings?

Gerben321, 2019-03-02:
I tried to match all settings (ser2net, dsmr_parser) to what works with cu, but I can't get it to work. I keep getting the same issue.

changed the settings to
`
SERIAL_SETTINGS_V5 = {
    'baudrate': 115200,
    'bytesize': serial.EIGHTBITS,
    'parity': serial.PARITY_NONE,
    'stopbits': serial.STOPBITS_ONE,
    'xonxoff': 1,
    'rtscts': 0,
    'timeout': 20
}
`
I made sure the xonxoff and rtscts matched, and now they do, but same result.

ndokter, 2019-03-02:
Hm thanks for trying. Im not sure yet what the issue is. It must be related to that reading part. I tried to parse your CU telegram data which parses correctly.

Gerben321, 2019-03-02:
Might have made a tiny bit of progress. In ser2net I've changed 8DATABITS to 7DATABITS and now when I look at the raw data via putty it looks exactly like what I get via cu! So that's good I think, but still nothing in Home Assistant. I've changed the databits in the settings.py file as well. I'll keep trying settings!

Gerben321, 2019-03-03:
Still no data in HASS, but I received a random error. Is it helpful?

2019-03-02 23:01:57 ERROR (MainThread) [dsmr_parser.clients.protocol] failed to parse telegram
Traceback (most recent call last):
  File "/usr/local/lib/python3.7/site-packages/dsmr_parser/clients/protocol.py", line 103, in handle_telegram
    parsed_telegram = self.telegram_parser.parse(telegram)
  File "/usr/local/lib/python3.7/site-packages/dsmr_parser/parsers.py", line 57, in parse
    telegram[signature] = parser.parse(match.group(0))
  File "/usr/local/lib/python3.7/site-packages/dsmr_parser/parsers.py", line 160, in parse
    return CosemObject(self._parse(line))
  File "/usr/local/lib/python3.7/site-packages/dsmr_parser/parsers.py", line 112, in _parse
    raise ParseError("Invalid '%s' line for '%s'", line, self)
dsmr_parser.exceptions.ParseError: ("Invalid '%s' line for '%s'", '1-0:2.8.2(000000>000*kWh)\r\n', <dsmr_parser.parsers.CosemParser object at 0x7f28208e8630>)
2019-03-03 02:13:08 ERROR (MainThread) [dsmr_parser.clients.protocol] failed to parse telegram
Traceback (most recent call last):
  File "/usr/local/lib/python3.7/site-packages/dsmr_parser/clients/protocol.py", line 103, in handle_telegram
    parsed_telegram = self.telegram_parser.parse(telegram)
  File "/usr/local/lib/python3.7/site-packages/dsmr_parser/parsers.py", line 57, in parse
    telegram[signature] = parser.parse(match.group(0))
  File "/usr/local/lib/python3.7/site-packages/dsmr_parser/parsers.py", line 160, in parse
    return CosemObject(self._parse(line))
  File "/usr/local/lib/python3.7/site-packages/dsmr_parser/parsers.py", line 112, in _parse
    raise ParseError("Invalid '%s' line for '%s'", line, self)
dsmr_parser.exceptions.ParseError: ("Invalid '%s' line for '%s'", '1-3:0.2.8(509\r\n', <dsmr_parser.parsers.CosemParser object at 0x7f28208e8240>)

ndokter, 2019-03-03:
That looks like something I can look into. I'll take a look to tomorrow

Gerben321, 2019-03-03:
Cheers!

ndokter, 2019-03-04:
When looking at your error i see that theres an unexpected `>` where it would expect a `.` `1-0:2.8.2(000000>000*kWh)\r\n`

Might still be related to the data not being read entirely correctly. Theres a slight chance that its valid (ill have to check the dsmr specs), but i doubt thats the case

Gerben321, 2019-03-04:
Ah that's weird. Can it have anything to do with parity or stopbits? I'm really new to this, so not sure about it all. A `.` indeed makes more sense at that spot...

ndokter, 2019-03-04:
Maybe i can make a workaround for you to replace these chars. Might not be the cleanest thing, but itl save you the trouble

Gerben321, 2019-03-04:
That would be great! I'll check the internet in the meanwhile. But almost nothing to find on DSMR v5 sadly...

ndokter, 2019-03-04:
See page 25 of the specification https://www.netbeheernederland.nl/_upload/Files/Slimme_meter_15_a727fce1f1.pdf

I also see issues in some of the other lines so a simple replace fix wont work :(
`1-0:1.8.1(000072.532:{wx9`
`0-1:24.2.1(190302112007w)(00028.843:}3)`

Gerben321, 2019-03-04:
Damn... I was checking that link indeed, but I don't understand how that is correct. When I use 8 databits I get complete scrambled data when I view via putty, but when I do it with 7 databits I get almost perfect data. All other settings do not matter. I'm a bit baffled...

ndokter, 2019-03-04:
Are the values consistently wrong? So each telegram results in the same errors? If not, then it might be a connection issue (bad cable or something)?

Gerben321, 2019-03-04:
It seems to be slightly different every time. I'll try to lend a USB P1 reader from someone to check if my current one isn't acting up. Thanks for all efforts so far!

ndokter, 2019-03-04:
On diffrent lines or the same line? If the same line changes slightly, then its less likely its an connection issue. Might still be worth checking out though

Gerben321, 2019-03-04:
On different lines, I've compared some telegrams, and some are good on a certain line, and others are not.

ndokter, 2019-03-04:
Alright thanks for checking. Then its worth trying a different cable or something to read it

Gerben321, 2019-03-04:
Working like a charm with an other cable! Got a P1 > USB cable now. No matter what I do, the P1 connector I used before doesn't work. Changing cables etc. also has no effect.

ndokter, 2019-03-05:
Ah, that's great! Glad you figured it out

closed 2019-03-05 by ndokter

0x00-0xFF, 2019-05-20:
I still got this error, tried a different P1 cable also...
Same error:
UnicodeDecodeError: 'ascii' codec can't decode byte 0xff in position 0: ordinal not in range(128)

Dyllaann, 2019-06-29:
I am also having problems, with the same error as posted by  @0x00-0xFF