ndokter/dsmr_parser #39: Parser not compatible with Belgian situation

joverbee, 2019-12-01:
I believe the parser does not correctly deal with a smart meter installed in Belgium (Fluvius netbeheerder). I am using the parser through the Home Assistant sensor.
For gas the telegram should be 0-1:24.2.3, but maybe there are other differences. I can help to provide test telegrams but need instructions on how to do so (totally new to home assistant and just learning to set it up).
Ideally a config switch for NL/BE would be great.

ndokter, 2019-12-01:
Thanks for reaching out. Im afraid that this will stay unsupported until someone wants to put in time to resolve this. I'm personally not that active with the project at this time.

Maybe someone else wants to pick it up and maybe im going to work on the project a little bit in the next 2 months, but im not promising anything :)

joverbee, 2019-12-02:
Ok, I understand.
For now I solved the situation by directly editing the obis_references.py changing 1 line to
GAS_METER_READING = r'\d-\d:24\.2\.3.+?\r\n'
inspired by  @dennissiemensma 
https://github.com/dennissiemensma/dsmr-reader/blob/master/dsmr_parser/obis_references.py
I have no clue at the moment on how to properly incorporate this into the dsmr_parser library and then to make it update in home assistant, let alone how to make an elegant configuration that depends on where you are.

ronniebee, 2019-12-12:
I get the foloowing message on my SSh:

Collecting dsmr-parser
  Using cached https://files.pythonhosted.org/packages/9e/d0/eb5c275d7b293f3fb6a754e017a79e3e929bf32c14e920e2a7278edfd04a/dsmr-parser-0.14.tar.gz
Collecting pyserial<4,>=3 (from dsmr-parser)
  Downloading https://files.pythonhosted.org/packages/0d/e4/2a744dd9e3be04a0c0907414e2a01a7c88bb3915cbe3c8cc06e209f59c30/pyserial-3.4-py2.py3-none-any.whl (193kB)
    100% |████████████████████████████████| 194kB 1.5MB/s 
Collecting pyserial-asyncio<1 (from dsmr-parser)
  Using cached https://files.pythonhosted.org/packages/41/3f/e26f71269cbc0890a527a736d9afc5c0d5838a2c188be680558d635b7dc2/pyserial-asyncio-0.4.tar.gz
    Complete output from command python setup.py egg_info:
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "/tmp/pip-build-6hhyQU/pyserial-asyncio/setup.py", line 15, in <module>
        raise RuntimeError("pyserial-asyncio requires at least Python 3.4")
    RuntimeError: pyserial-asyncio requires at least Python 3.4
    
    ----------------------------------------
**Command "python setup.py egg_info" failed with error code 1 in /tmp/pip-build-6hhyQU/pyserial-asyncio**/

What do I need to do to resolve this? (yes I'm a newbee ... :-) )

ndokter, 2019-12-21:
@dupondje has added Belgium meter support. Does this also cover your request?

https://github.com/ndokter/dsmr_parser/pull/44

joverbee, 2019-12-22:
Yes, this is a much cleaner way than my hard-coded attempt. Thanks @dupondje.

closed 2019-12-22 by joverbee