ndokter/dsmr_parser #41: What to do if i do not see any readings

MHolkamp, 2019-12-10:
Hi,

I have the following in my configuration.yaml

sensor:
  - platform: dsmr
    port: /dev/ttyUSB1
    dsmr_version: 4.0

group: 
   # !include groups.yaml
    
  meter_readings:
    name: Meter Waardes
    entities:
      - sensor.power_consumption_low
      - sensor.power_consumption_normal
      - sensor.power_production_low
      - sensor.power_production_normal
      - sensor.gas_consumption

But when i add something to my gui is see:
unknown

ndokter, 2019-12-12:
Hey, i dont know what integration you are using. The client to read the serial port might not connect using the right settings. They are defined here:

https://github.com/ndokter/dsmr_parser/blob/master/dsmr_parser/clients/settings.py

ronniebee, 2019-12-19:
Hi, ik heb hetzelfde probleem. Als ik de parser installeer krijg ik:

Pakketlijsten worden ingelezen... Klaar
Boom van vereisten wordt opgebouwd       
De statusinformatie wordt gelezen... Klaar
E: Kan pakket dsmr-parser niet vinden

Wat het laatste betekent weet ik niet.

Mijn log in portainer (ik gerbuik docker) geeft aan:

2019-12-19 12:31:23 INFO (SyncWorker_1) [homeassistant.loader] Loaded config from homeassistant.components.config,
2019-12-19 12:31:23 INFO (SyncWorker_8) [homeassistant.loader] Loaded frontend from homeassistant.components.frontend,
2019-12-19 12:31:23 INFO (SyncWorker_5) [homeassistant.loader] Loaded api from homeassistant.components.api,
2019-12-19 12:31:23 INFO (SyncWorker_9) [homeassistant.loader] Loaded auth from homeassistant.components.auth,
2019-12-19 12:31:23 INFO (SyncWorker_3) [homeassistant.loader] Loaded lovelace from homeassistant.components.lovelace,
2019-12-19 12:31:23 INFO (SyncWorker_2) [homeassistant.loader] Loaded onboarding from homeassistant.components.onboarding,
2019-12-19 12:31:23 INFO (SyncWorker_4) [homeassistant.loader] Loaded system_log from homeassistant.components.system_log,
2019-12-19 12:31:23 INFO (SyncWorker_7) [homeassistant.loader] Loaded websocket_api from homeassistant.components.websocket_api,
2019-12-19 12:31:23 INFO (SyncWorker_6) [homeassistant.loader] Loaded history from homeassistant.components.history,
2019-12-19 12:31:23 INFO (SyncWorker_0) [homeassistant.loader] Loaded recorder from homeassistant.components.recorder,
2019-12-19 12:31:23 INFO (SyncWorker_1) [homeassistant.loader] Loaded logbook from homeassistant.components.logbook,
2019-12-19 12:31:23 INFO (SyncWorker_8) [homeassistant.loader] Loaded map from homeassistant.components.map,
2019-12-19 12:31:23 INFO (SyncWorker_5) [homeassistant.loader] Loaded mobile_app from homeassistant.components.mobile_app,
2019-12-19 12:31:23 INFO (SyncWorker_9) [homeassistant.loader] Loaded person from homeassistant.components.person,
2019-12-19 12:31:23 INFO (SyncWorker_3) [homeassistant.loader] Loaded ssdp from homeassistant.components.ssdp,
2019-12-19 12:31:23 INFO (SyncWorker_2) [homeassistant.loader] Loaded sun from homeassistant.components.sun,
2019-12-19 12:31:23 INFO (SyncWorker_4) [homeassistant.loader] Loaded system_health from homeassistant.components.system_health,
2019-12-19 12:31:23 INFO (SyncWorker_7) [homeassistant.loader] Loaded updater from homeassistant.components.updater,
2019-12-19 12:31:23 INFO (SyncWorker_6) [homeassistant.loader] Loaded zeroconf from homeassistant.components.zeroconf,
2019-12-19 12:31:23 INFO (MainThread) [homeassistant.bootstrap] Setting up {'system_log'},
2019-12-19 12:31:23 INFO (MainThread) [homeassistant.setup] Setting up http,
2019-12-19 12:31:23 INFO (MainThread) [homeassistant.setup] Setup of domain http took 0.0 seconds.,
2019-12-19 12:31:23 INFO (MainThread) [homeassistant.setup] Setting up system_log,
2019-12-19 12:31:23 INFO (MainThread) [homeassistant.setup] Setup of domain system_log took 0.0 seconds.,
2019-12-19 12:31:24 INFO (MainThread) [homeassistant.setup] Setting up recorder,
2019-12-19 12:31:24 INFO (MainThread) [homeassistant.setup] Setting up webhook,
2019-12-19 12:31:24 INFO (MainThread) [homeassistant.setup] Setup of domain webhook took 0.0 seconds.,
2019-12-19 12:31:27 INFO (MainThread) [homeassistant.setup] Setting up cloud,
2019-12-19 12:31:27 INFO (MainThread) [homeassistant.setup] Setup of domain recorder took 3.2 seconds.,
2019-12-19 12:31:27 INFO (MainThread) [homeassistant.setup] Setup of domain cloud took 0.0 seconds.,
2019-12-19 12:31:27 INFO (MainThread) [homeassistant.setup] Setting up sun,
2019-12-19 12:31:28 INFO (MainThread) [homeassistant.setup] Setup of domain sun took 0.2 seconds.,
2019-12-19 12:31:28 INFO (MainThread) [homeassistant.setup] Setting up person,
2019-12-19 12:31:28 INFO (MainThread) [homeassistant.setup] Setting up lovelace,
2019-12-19 12:31:28 INFO (MainThread) [homeassistant.setup] Setup of domain lovelace took 0.0 seconds.,
2019-12-19 12:31:28 INFO (MainThread) [homeassistant.setup] Setting up updater,
2019-12-19 12:31:28 INFO (MainThread) [homeassistant.setup] Setup of domain updater took 0.0 seconds.,
2019-12-19 12:31:28 INFO (MainThread) [homeassistant.setup] Setting up group,
2019-12-19 12:31:28 INFO (MainThread) [homeassistant.setup] Setup of domain group took 0.0 seconds.,
2019-12-19 12:31:28 INFO (MainThread) [homeassistant.setup] Setting up ssdp,
2019-12-19 12:31:28 INFO (MainThread) [homeassistant.setup] Setup of domain ssdp took 0.0 seconds.,
2019-12-19 12:31:28 INFO (MainThread) [homeassistant.setup] Setting up scene,
2019-12-19 12:31:28 INFO (MainThread) [homeassistant.components.scene] Setting up scene.homeassistant,
2019-12-19 12:31:28 INFO (SyncWorker_6) [homeassistant.loader] Loaded binary_sensor from homeassistant.components.binary_sensor,
2019-12-19 12:31:28 INFO (SyncWorker_0) [homeassistant.loader] Loaded google_translate from homeassistant.components.google_translate,
2019-12-19 12:31:28 INFO (MainThread) [homeassistant.setup] Setting up auth,
2019-12-19 12:31:28 INFO (MainThread) [homeassistant.setup] Setup of domain auth took 0.0 seconds.,
2019-12-19 12:31:29 INFO (MainThread) [homeassistant.setup] Setting up mobile_app,
2019-12-19 12:31:29 INFO (MainThread) [homeassistant.setup] Setting up history,
2019-12-19 12:31:29 INFO (MainThread) [homeassistant.setup] Setup of domain history took 0.0 seconds.,
[... cut ...]
2019-12-19 12:31:35 ERROR (MainThread) [homeassistant.components.http] Failed to create HTTP server at port 8123: [Errno 98] error while attempting to bind on address ('0.0.0.0', 8123): address in use,

Deze laatste error weet ik niet op te lossen. Wat dien ik hier te doen (nb ik gebruik duckdns..)? 
mijn hassio draait nl  op 8123.

ben al > 30 uur bezig dsmr uit te lezen maar geen resultaat.. 

mijn config.yaml ziet er als volgt uit.

sensor:
  - platform: dsmr_reader 
  - platform: dsmr
    port: /dev/ttyUSB0
    dsmr_version: 5

Kan je  mij verder helpen? Ik ben een newbee en heb al veel aan de praat gekregen maar dit lijkt een brug te ver..

Gr.

ndokter, 2019-12-21:
Hi i can't help you with that sorry. It's not directly related to this project and has to do with homeasisstant

It looks like it starts incorrectly after which port 8123 is already used. My best guess is to check which app is using that port using netcat or something. Or maybe youre already running homeassistant as a daemon process

closed 2019-12-21 by ndokter