ndokter/dsmr_parser #50: Limit information-overflow (to HASS database)

tbressers, 2020-02-28:
Feature request

Would it be possible to add a configuration item to reduce the maximum generated amount of data per minute (or hour)? My home-assistant database is flooded, and there seems no good way to prevent that (hence, the recorder purging doesn't seem to do much / enough)

ndokter, 2020-02-28:
I should check the implementation in home-assistent. This change makes a little bit more sense in HA at first glance.

tbressers, 2020-02-28:
Thank you... I will do so

ndokter, 2020-02-28:
I could also be a feature in the serial reader though, but im not sure if HA uses that (i dont think so)

tbressers, 2020-02-28:
Somehow it does feel like something one would like to solve as low-level as
possible... but I don't know the architecture enough to determine where
such an implementation would fit best.

ndokter, 2020-03-01:
Yeah it should be implemented somewhere in https://github.com/home-assistant/home-assistant/blob/dev/homeassistant/components/dsmr/sensor.py

closed 2020-03-01 by ndokter