ndokter/dsmr_parser #56: Rename project to Digital Smart Meter Requirements since you seem to support more than only Dutch Smart Meter telegrams

jmartens, 2020-09-05:
It seems you are supporting the Benelux region now and not only the dutch smart meter protocol, shouldn't the naming of the project reflect that?

To keep the D I would suggest to change it to Digital Smart Meter Requirements instead of Dutch Smart Meter Requirements, what do you think?

RobBie1221, 2020-09-24:
Wouldn't that make it confusing? The readme says:
A library for parsing Dutch Smart Meter Requirements (DSMR) telegram data.
It is doing exactly that, parsing data in format defined by "Netbeheer Nederland" written down in a document called "Dutch Smart Meter Requirements":
https://www.netbeheernederland.nl/_upload/Files/Slimme_meter_15_9e94f87c14.pdf

It seems to me that Belgium chose to comply (with some minor tweaks) to the Dutch standard.

ndokter, 2020-10-26:
Agree'd with @RobBie1221 that the name seems ok for now

closed 2020-10-26 by ndokter