ndokter/dsmr_parser #60: Parser value error '2C'

AussiSG, 2020-10-22:
It seems sometimes when the value in the gasreading returns "2C" the parser can't handle this.

Is this something i've wrongly setup in Home assistant?

Note:  I am running the DMSR add-on on Home Assistant with a Landys & Gyr E350, however with dsmr 2.2.

```
2020-10-22 11:00:09 DEBUG (MainThread) [dsmr_parser.clients.protocol] got telegram: /XMX5XMXABCE000069683

0-0:96.1.1(31333637343139322020202020202020)
1-0:1.8.1(15237.388*kWh)
1-0:1.8.2(18759.401*kWh)
1-0:2.8.1(00000.000*kWh)
1-0:2.8.2(00000.000*kWh)
0-0:96.14.0(0002)
1-0:1.7.0(0002.94*kW)
1-0:2.7.0(0000.00*kW)
0-0:96.13.1()
0-0:96.13.0()
0-1:96.1.0(3238303131303038333032323537303133)
0-1:24.1.0(03)
0-1:24.3.0(201022100000)(2C)(60)(1)(0-1:24.2.0)(m3)
(03575.334)
!

2020-10-22 11:00:09 ERROR (MainThread) [homeassistant] Error doing job: Fatal error: protocol.data_received() call failed.
Traceback (most recent call last):
  File "/usr/local/lib/python3.8/asyncio/selector_events.py", line 860, in _read_ready__data_received
    self._protocol.data_received(data)
  File "/usr/local/lib/python3.8/site-packages/dsmr_parser/clients/protocol.py", line 92, in data_received
    self.handle_telegram(telegram)
  File "/usr/local/lib/python3.8/site-packages/dsmr_parser/clients/protocol.py", line 107, in handle_telegram
    parsed_telegram = self.telegram_parser.parse(telegram)
  File "/usr/local/lib/python3.8/site-packages/dsmr_parser/parsers.py", line 57, in parse
    telegram[signature] = parser.parse(match.group(0))
  File "/usr/local/lib/python3.8/site-packages/dsmr_parser/parsers.py", line 159, in parse
    return MBusObject(self._parse(line))
  File "/usr/local/lib/python3.8/site-packages/dsmr_parser/parsers.py", line 137, in _parse
    return [self.value_formats[i].parse(value)
  File "/usr/local/lib/python3.8/site-packages/dsmr_parser/parsers.py", line 137, in <listcomp>
    return [self.value_formats[i].parse(value)
  File "/usr/local/lib/python3.8/site-packages/dsmr_parser/parsers.py", line 235, in parse
    value = self.coerce_type(value) if value is not None else value
ValueError: invalid literal for int() with base 10: '2C'
2020-10-22 11:00:09 ERROR (MainThread) [dsmr_parser.clients.protocol] disconnected due to exception
NoneType: None
```

ndokter, 2020-10-26:
This seems like a valid bug where there is something in the DSMR specification that the parser cannot deal with correctly yet

AussiSG, 2020-10-26:
Found this one while "googling" ;-)

https://github.com/dennissiemensma/dsmr-reader/issues/458

Seems they resolved it here ?

ndokter, 2020-10-26:
Ah great thanks! Not sure when and if ill fix it, but i'll use that if i do!

AussiSG, 2020-10-26:
I hope you do, it kills my gas readings every once and a while ;-)

lowdef, 2020-11-01:
See pull request mentioned above, solves your issue.
(pull request https://github.com/ndokter/dsmr_parser/pull/61)

Note that in 
```
0-1:24.3.0(201022100000)(2C)(60)(1)(0-1:24.2.0)(m3)
(03575.334)
```

I could not find any specification of what the values `(2C)(60)(1)` are actually used for. Also while the parser parses them, the values are normally basically ignored as far as I could see. So changes the parsing type seems to be harmless.

Tox tests run 100%.
I did not add a test for this particular case.

Following code demonstrates that the issue is fixed:
```python
#!/usr/bin/env python3
# -*- coding: utf-8 -*-
"""
Created on Sun May  3 12:27:42 2020

@author: lowdef
"""

from dsmr_parser import telegram_specifications
from dsmr_parser.objects import Telegram
from dsmr_parser.parsers import TelegramParser
from dsmr_parser.parsers import MBusParser
from dsmr_parser.parsers import ValueParser
from dsmr_parser import obis_references
from dsmr_parser.value_types import timestamp
from decimal import Decimal

telegram_issue60  = r"""-0:96.1.1(31333637343139322020202020202020)
1-0:1.8.1(15237.388*kWh)
1-0:1.8.2(18759.401*kWh)
1-0:2.8.1(00000.000*kWh)
1-0:2.8.2(00000.000*kWh)
0-0:96.14.0(0002)
1-0:1.7.0(0002.94*kW)
1-0:2.7.0(0000.00*kW)
0-0:96.13.1()
0-0:96.13.0()
0-1:96.1.0(3238303131303038333032323537303133)
0-1:24.1.0(03)
0-1:24.3.0(201022100000)(2C)(60)(1)(0-1:24.2.0)(m3)
(03575.334)
!
"""

sample = telegram_issue60.replace('\n', '\r\n')

telegram_specification = telegram_specifications.V2_2
telegram_specification['checksum_support']=False

parser = TelegramParser(telegram_specification)
telegram = Telegram(sample, parser, telegram_specification)

print(telegram)

```

Demonstrates that the code works:
```
>>> runfile('/home/.../projects/donderstraal/develop/issue60.py', wdir='/home/.../projects/donderstraal/develop')
ELECTRICITY_USED_TARIFF_1: 	 15237.388	[kWh]
ELECTRICITY_USED_TARIFF_2: 	 18759.401	[kWh]
ELECTRICITY_DELIVERED_TARIFF_1: 	 0.000	[kWh]
ELECTRICITY_DELIVERED_TARIFF_2: 	 0.000	[kWh]
ELECTRICITY_ACTIVE_TARIFF: 	 0002	[None]
CURRENT_ELECTRICITY_USAGE: 	 2.94	[kW]
CURRENT_ELECTRICITY_DELIVERY: 	 0.00	[kW]
TEXT_MESSAGE_CODE: 	 None	[None]
TEXT_MESSAGE: 	 None	[None]
EQUIPMENT_IDENTIFIER_GAS: 	 3238303131303038333032323537303133	[None]
DEVICE_TYPE: 	 03	[None]
GAS_METER_READING: 	 3575.334	[m3] at 2020-10-22T10:00:00+02:00
Process finished with exit code 0

```

lowdef, 2020-12-16:
I believe this thread can be closed, the issue was resolved in version 0.23

closed 2023-05-17 by AussiSG