ndokter/dsmr_parser #65: HA: ValueError: invalid literal for int() with base 16: '140W'

matthijsberg, 2020-12-12:
HI, 

I'm using this in Home Assistant. I get quite some CRC errors (`2020-12-11 21:01:13 WARNING (MainThread) [dsmr_parser.clients.protocol] Invalid telegram. The CRC checksum '45204' does not match the expected '29681'`) and the error below. I connect to a Landis & Gyr E360 via DSMR 5 directly connected. It looks like I get a value in the CRC field? most of the error are CRC, sometimes I get this:

```
2020-12-12 14:11:47 ERROR (MainThread) [homeassistant] Error doing job: Exception in callback SerialTransport._read_ready()
Traceback (most recent call last):
  File "/usr/local/lib/python3.8/asyncio/events.py", line 81, in _run
    self._context.run(self._callback, *self._args)
  File "/usr/local/lib/python3.8/site-packages/serial_asyncio/__init__.py", line 106, in _read_ready
    self._protocol.data_received(data)
  File "/usr/local/lib/python3.8/site-packages/dsmr_parser/clients/protocol.py", line 92, in data_received
    self.handle_telegram(telegram)
  File "/usr/local/lib/python3.8/site-packages/dsmr_parser/clients/protocol.py", line 107, in handle_telegram
    parsed_telegram = self.telegram_parser.parse(telegram)
  File "/usr/local/lib/python3.8/site-packages/dsmr_parser/parsers.py", line 47, in parse
    self.validate_checksum(telegram_data)
  File "/usr/local/lib/python3.8/site-packages/dsmr_parser/parsers.py", line 83, in validate_checksum
    expected_crc = int(checksum_hex.group(0), base=16)
ValueError: invalid literal for int() with base 16: '140W'
```

Let me know if you need more information (and some pointers where to get it) if needed.

ndokter, 2022-01-04:
What would help if you could add raw telegram data. User lowdef explained how that could be done here:
https://github.com/ndokter/dsmr_parser/issues/91#issuecomment-962428443

dupondje, 2023-11-09:
Closed as no response received.

closed 2023-11-09 by dupondje