ndokter/dsmr_parser #67: HA: parser error for 99.97.0

didiermiller, 2020-12-14:
Hi,

I am running into the following error. For some reason the telegram information for OBIS 99.97.0 is empty for my ISKRA-AM550. I assume this has always been the case, I am not sure why that would change suddenly.

I upgraded to the latest version of HA (2020.12.0 and OS 5.8) and suddenly this because an issue.
`
Logger: dsmr_parser.clients.protocol
Source: /usr/local/lib/python3.8/site-packages/dsmr_parser/clients/protocol.py:112
First occurred: 10:56:35 AM (394 occurrences)
Last logged: 11:03:39 AM
failed to parse telegram

Traceback (most recent call last):
  File "/usr/local/lib/python3.8/site-packages/dsmr_parser/clients/protocol.py", line 108, in handle_telegram
    parsed_telegram = self.telegram_parser.parse(telegram)
  File "/usr/local/lib/python3.8/site-packages/dsmr_parser/parsers.py", line 59, in parse
    telegram[signature] = parser.parse(match.group(0))
  File "/usr/local/lib/python3.8/site-packages/dsmr_parser/parsers.py", line 249, in parse
    return ProfileGenericObject(self._parse(line))
  File "/usr/local/lib/python3.8/site-packages/dsmr_parser/parsers.py", line 148, in _parse
    raise ParseError("Invalid '%s' line for '%s'", line, self)
dsmr_parser.exceptions.ParseError: ("Invalid '%s' line for '%s'", '99.97.0()\r\n', <dsmr_parser.parsers.ProfileGenericParser object at 0x7f13411fe2e0>)
`
I removed all entries for 99.97 in the python code (power event log or something along those lines) and that helped.

Here is an example of the actual telegram:
`
1-3:0.2.8(50)
0-0:1.0.0(201214112909W)
0-0:96.1.1(4530303533303037353234343134343139)
1-0:1.8.1(000177.571*kWh)
1-0:1.8.2(000130.312*kWh)
1-0:2.8.1(000072.503*kWh)
1-0:2.8.2(000031.774*kWh)
0-0:96.14.0(0002)
1-0:1.7.0(00.000*kW)
1-0:2.7.0(00.139*kW)
0-0:96.7.21(00010)
0-0:96.7.9(00002)
1-0:99.97.0()
1-0:32.32.0(00000)
1-0:32.36.0(00002)
0-0:96.13.0()
1-0:32.7.0(241.2*V)
1-0:31.7.0(001*A)
1-0:21.7.0(00.000*kW)
1-0:22.7.0(00.134*kW)
!72C0
~/ISK5\2M550E-1013
`

fnxpt, 2020-12-14:
having the same issue with DSMR 5

ndokter, 2020-12-14:
Might be because of recent changes that touched this code:

**ProfileGeneric parser working, TODO complete ProfileGenericObject + Test**
https://github.com/ndokter/dsmr_parser/commit/b6278a8991c8729c22c35cc0960fa0ff1dc72518

**make all objects able to print their own values**
https://github.com/ndokter/dsmr_parser/commit/a0ce89054a02651c86fe8a4559a4218a5047df5c

juliusgaiuscaesar, 2020-12-14:
I can confirm the issue with the ISKRA-AM550 on HA after the upgrade to v2020.12.0 for HA and v5.8 for the OS.

fnxpt, 2020-12-14:
looks like there is a PR already #57

ndokter, 2020-12-14:
Thanks for bringing up #57. I've merged the supposed fix and created a new build. Is anyone able to test it? It's released as version 0.25

fnxpt, 2020-12-14:
the issue is still there 
```Traceback (most recent call last):
  File "/usr/local/lib/python3.8/site-packages/dsmr_parser/clients/protocol.py", line 108, in handle_telegram
    parsed_telegram = self.telegram_parser.parse(telegram)
  File "/usr/local/lib/python3.8/site-packages/dsmr_parser/parsers.py", line 59, in parse
    telegram[signature] = parser.parse(match.group(0))
  File "/usr/local/lib/python3.8/site-packages/dsmr_parser/parsers.py", line 249, in parse
    return ProfileGenericObject(self._parse(line))
  File "/usr/local/lib/python3.8/site-packages/dsmr_parser/parsers.py", line 148, in _parse
    raise ParseError("Invalid '%s' line for '%s'", line, self)
dsmr_parser.exceptions.ParseError: ("Invalid '%s' line for '%s'", '99.97.0()\r\n', <dsmr_parser.parsers.ProfileGenericParser object at 0x7ff653e298b0>)```

fnxpt, 2020-12-14:
It looks like I was running a older version... running dsmr_console I get the following error
```AttributeError: 'ProfileGenericObject' object has no attribute 'value'```

lowdef, 2020-12-15:
The fix for https://github.com/ndokter/dsmr_parser/pull/57 is erroneous as also already indicated in that thread. Not sure why it is merged now? Please revert this merge.

What is causing this is that several meters are not conforming to the standard, alas we can not fix that.

The solution is to make the parser robust against such culprits.

I also sketched the solution in the comments to issue https://github.com/ndokter/dsmr_parser/pull/57: https://github.com/ndokter/dsmr_parser/pull/57#pullrequestreview-536008205.

We need to recognize such cases and return a valid ProfileGeneric Object.

I suggest you rewrite, such that created GenericParserObject has valid entries and does not break. I think what would work is if the following values are returned for this special case:

A valid ProfileGeneric object representing an empty value has the following member values:
```
buffer_type : None
buffer_length : 0
buffer : []
values : []
```

didiermiller, 2020-12-15:
I agree, this should be a None object. Upvoted your comment

ndokter, 2020-12-15:
I've reverted the fix that was merged yesterday (#57), so we're back at the original issue

RobBie1221, 2020-12-16:
I've pinned 0.25 to Home Assistant, at least from that side it fixes the issue. The empty value represents `POWER_EVENT_FAILURE_LOG` which is not parsed from HA side, so it won't bother that the ProfileGeneric object is invalid.

I'll see if I can push the solution proposed by @lowdef to have a sustainable solution that works for everyone.

lowdef, 2020-12-16:
@RobBie1221 are you going to provide a code change, or shall I have a go at it during the Christmas vacation?

fnxpt, 2020-12-16:
I can confirm that now it works on Home Assistant, thank you guys.

RobBie1221, 2020-12-16:
@lowdef I was planning to provide a code change but if you are willing to have a look at it then that's fine by me as well. You seem to have a better understanding of how it should work, I'll have to dig into that :-)

lowdef, 2020-12-24:
Pull Request #69 provides a fix for the issue.

closed 2023-11-09 by dupondje