ndokter/dsmr_parser #75: UnicodeDecodeError with Tasmota Serial to TCP Bridge

nagyrobi, 2021-02-18:
Using a Wemos D1 Mini module to connect with a SmartMeter, running Tasmota firmware forwarding data in through [Serial to TCP Bridge](https://tasmota.github.io/docs/Serial-to-TCP-Bridge/).

Although in Home Assistant it is possible to connect with serial 2 network proxies, this fails for me with the following error:
```
021-02-18 17:56:16 ERROR (MainThread) [homeassistant] Error doing job: Fatal error: protocol.data_received() call failed.
Traceback (most recent call last):
  File "/usr/local/lib/python3.8/asyncio/selector_events.py", line 862, in _read_ready__data_received
    self._protocol.data_received(data)
  File "/usr/local/lib/python3.8/site-packages/dsmr_parser/clients/protocol.py", line 91, in data_received
    data = data.decode('ascii')
UnicodeDecodeError: 'ascii' codec can't decode byte 0xff in position 2: ordinal not in range(128)
2021-02-18 17:56:16 ERROR (MainThread) [dsmr_parser.clients.protocol] disconnected due to exception
Traceback (most recent call last):
  File "/usr/local/lib/python3.8/asyncio/selector_events.py", line 862, in _read_ready__data_received
    self._protocol.data_received(data)
  File "/usr/local/lib/python3.8/site-packages/dsmr_parser/clients/protocol.py", line 91, in data_received
    data = data.decode('ascii')
UnicodeDecodeError: 'ascii' codec can't decode byte 0xff in position 2: ordinal not in range(128)
```
Could this pull request be a solution for me? https://github.com/ndokter/dsmr_parser/pull/70

Aeroid, 2021-11-21:
@nagyrobi try v0.31 #92 which uses latin-1 instead of ascii

nagyrobi, 2021-11-21:
Thanks but moved along since, as I was able to do it in [another way](https://tasmota.github.io/docs/Smart-Meter-Interface/#sanxing-sx6x1-sxxu1x-ascii-obis).

closed 2021-11-21 by nagyrobi