ndokter/dsmr_parser #79: No support for DSMR 4.2.2

nihad-mehmedovic, 2021-05-09:
I have a Kaifa MA304C which is using dsmr 4.4.2 but that doesn't seem parsed. I'm using it in combination with https://www.home-assistant.io/integrations/dsmr/

Are you gonna support this in a near feature?

sander76, 2021-08-21:
I think I have something similar. I am using the "slimme meter" integration from home assistant, which uses this under the hood. Setting my KAIFA MA105c (which is DSMR 4.2.2) as a DSMR 4 device does not seem to parse "power consumption [kW]". No idea if this library is to blame though.
But as you requested, please see below the incoming raw data:

```
/KFM5KAIFA-METER

1-3:0.2.8(42)
0-0:1.0.0(210821160713S)
0-0:96.1.1(4530303235303030303336303537373135)
1-0:1.8.1(005921.506*kWh)
1-0:1.8.2(008182.550*kWh)
1-0:2.8.1(000724.587*kWh)
1-0:2.8.2(001592.952*kWh)
0-0:96.14.0(0001)
1-0:1.7.0(00.000*kW)
1-0:2.7.0(00.440*kW)
0-0:96.7.21(00016)
0-0:96.7.9(00012)
1-0:99.97.0(9)(0-0:96.7.19)(190602053824S)(0000004877*s)(190418170516S)(0000000264*s)(190418170037S)(0000000656*s)(190201235252W)(0000003234*s)(181122133445W)(0000002849*s)(161117115736W)(0000006548*s)(160510105344S)(0000004203*s)(160414151532S)(0000007011*s)(000101000001W)(2147483647*s)
1-0:32.32.0(00001)
1-0:32.36.0(00000)
0-0:96.13.1()
0-0:96.13.0()
1-0:31.7.0(002*A)
1-0:21.7.0(00.000*kW)
1-0:22.7.0(00.440*kW)
0-1:24.1.0(003)
0-1:96.1.0(4730303332353631323434393036373135)
0-1:24.2.1(210821160000S)(06655.682*m3)
!AE43
/KFM5KAIFA-METER

1-3:0.2.8(42)
0-0:1.0.0(210821160722S)
0-0:96.1.1(4530303235303030303336303537373135)
1-0:1.8.1(005921.506*kWh)
1-0:1.8.2(008182.550*kWh)
1-0:2.8.1(000724.588*kWh)
1-0:2.8.2(001592.952*kWh)
0-0:96.14.0(0001)
1-0:1.7.0(00.000*kW)
1-0:2.7.0(00.446*kW)
0-0:96.7.21(00016)
0-0:96.7.9(00012)
1-0:99.97.0(9)(0-0:96.7.19)(190602053824S)(0000004877*s)(190418170516S)(0000000264*s)(190418170037S)(0000000656*s)(190201235252W)(0000003234*s)(181122133445W)(0000002849*s)(161117115736W)(0000006548*s)(160510105344S)(0000004203*s)(160414151532S)(0000007011*s)(000101000001W)(2147483647*s)
1-0:32.32.0(00001)
1-0:32.36.0(00000)
0-0:96.13.1()
0-0:96.13.0()
1-0:31.7.0(002*A)
1-0:21.7.0(00.000*kW)
1-0:22.7.0(00.446*kW)
0-1:24.1.0(003)
0-1:96.1.0(4730303332353631323434393036373135)
0-1:24.2.1(210821160000S)(06655.682*m3)
!9155
/KFM5KAIFA-METER

1-3:0.2.8(42)
0-0:1.0.0(210821160732S)
0-0:96.1.1(4530303235303030303336303537373135)
1-0:1.8.1(005921.506*kWh)
1-0:1.8.2(008182.550*kWh)
1-0:2.8.1(000724.590*kWh)
1-0:2.8.2(001592.952*kWh)
0-0:96.14.0(0001)
1-0:1.7.0(00.000*kW)
1-0:2.7.0(00.451*kW)
0-0:96.7.21(00016)
0-0:96.7.9(00012)
1-0:99.97.0(9)(0-0:96.7.19)(190602053824S)(0000004877*s)(190418170516S)(0000000264*s)(190418170037S)(0000000656*s)(190201235252W)(0000003234*s)(181122133445W)(0000002849*s)(161117115736W)(0000006548*s)(160510105344S)(0000004203*s)(160414151532S)(0000007011*s)(000101000001W)(2147483647*s)
1-0:32.32.0(00001)
1-0:32.36.0(00000)
0-0:96.13.1()
0-0:96.13.0()
1-0:31.7.0(002*A)
1-0:21.7.0(00.000*kW)
1-0:22.7.0(00.452*kW)
0-1:24.1.0(003)
0-1:96.1.0(4730303332353631323434393036373135)
0-1:24.2.1(210821160000S)(06655.682*m3)
!8E7B

```

lowdef, 2021-09-03:
hi @sander76,

when I parse one of you telegrams  (the first) with the dsmr_parser,  we get  a perfect result:
```console
P1_MESSAGE_HEADER: 	 42	[None]
P1_MESSAGE_TIMESTAMP: 	 2021-08-21T16:07:13+02:00	[None]
EQUIPMENT_IDENTIFIER: 	 4530303235303030303336303537373135	[None]
ELECTRICITY_USED_TARIFF_1: 	 5921.506	[kWh]
ELECTRICITY_USED_TARIFF_2: 	 8182.550	[kWh]
ELECTRICITY_DELIVERED_TARIFF_1: 	 724.587	[kWh]
ELECTRICITY_DELIVERED_TARIFF_2: 	 1592.952	[kWh]
ELECTRICITY_ACTIVE_TARIFF: 	 0001	[None]
CURRENT_ELECTRICITY_USAGE: 	 0.000	[kW]
CURRENT_ELECTRICITY_DELIVERY: 	 0.440	[kW]
SHORT_POWER_FAILURE_COUNT: 	 16	[None]
LONG_POWER_FAILURE_COUNT: 	 12	[None]
POWER_EVENT_FAILURE_LOG: 	 	 buffer length: 9
	 buffer type: 0-0:96.7.19
	 event occured at: 2019-06-02T05:38:24+02:00	 for: 4877 [s]
	 event occured at: 2019-04-18T17:05:16+02:00	 for: 264 [s]
	 event occured at: 2019-04-18T17:00:37+02:00	 for: 656 [s]
	 event occured at: 2019-02-01T23:52:52+01:00	 for: 3234 [s]
	 event occured at: 2018-11-22T13:34:45+01:00	 for: 2849 [s]
	 event occured at: 2016-11-17T11:57:36+01:00	 for: 6548 [s]
	 event occured at: 2016-05-10T10:53:44+02:00	 for: 4203 [s]
	 event occured at: 2016-04-14T15:15:32+02:00	 for: 7011 [s]
	 event occured at: 2000-01-01T00:00:01+01:00	 for: 2147483647 [s]
VOLTAGE_SAG_L1_COUNT: 	 1	[None]
VOLTAGE_SWELL_L1_COUNT: 	 0	[None]
TEXT_MESSAGE_CODE: 	 None	[None]
TEXT_MESSAGE: 	 None	[None]
DEVICE_TYPE: 	 3	[None]
INSTANTANEOUS_CURRENT_L1: 	 2	[A]
INSTANTANEOUS_ACTIVE_POWER_L1_POSITIVE: 	 0.000	[kW]
INSTANTANEOUS_ACTIVE_POWER_L1_NEGATIVE: 	 0.440	[kW]
EQUIPMENT_IDENTIFIER_GAS: 	 4730303332353631323434393036373135	[None]
HOURLY_GAS_METER_READING: 	 6655.682	[m3] at 2021-08-21T16:00:00+02:00
```

You're not using any power, because you are delivering electricity back to the network. (Je wekt meer op dan je gebruikt ;-)

```console
CURRENT_ELECTRICITY_USAGE: 	 0.000	[kW]
CURRENT_ELECTRICITY_DELIVERY: 	 0.440	[kW]
```

You can parse telegrams yourself with the following code:
```python
#!/usr/bin/env python3
# -*- coding: utf-8 -*-
"""
Created on Fri Sep 3 22:25:00 2021

@author: lowdef
"""

from dsmr_parser import telegram_specifications
from dsmr_parser.objects import Telegram
from dsmr_parser.parsers import TelegramParser

telegram_issue  = r"""/KFM5KAIFA-METER

1-3:0.2.8(42)
0-0:1.0.0(210821160713S)
0-0:96.1.1(4530303235303030303336303537373135)
1-0:1.8.1(005921.506*kWh)
1-0:1.8.2(008182.550*kWh)
1-0:2.8.1(000724.587*kWh)
1-0:2.8.2(001592.952*kWh)
0-0:96.14.0(0001)
1-0:1.7.0(00.000*kW)
1-0:2.7.0(00.440*kW)
0-0:96.7.21(00016)
0-0:96.7.9(00012)
1-0:99.97.0(9)(0-0:96.7.19)(190602053824S)(0000004877*s)(190418170516S)(0000000264*s)(190418170037S)(0000000656*s)(190201235252W)(0000003234*s)(181122133445W)(0000002849*s)(161117115736W)(0000006548*s)(160510105344S)(0000004203*s)(160414151532S)(0000007011*s)(000101000001W)(2147483647*s)
1-0:32.32.0(00001)
1-0:32.36.0(00000)
0-0:96.13.1()
0-0:96.13.0()
1-0:31.7.0(002*A)
1-0:21.7.0(00.000*kW)
1-0:22.7.0(00.440*kW)
0-1:24.1.0(003)
0-1:96.1.0(4730303332353631323434393036373135)
0-1:24.2.1(210821160000S)(06655.682*m3)
!AE43
"""

sample = telegram_issue.replace('\n', '\r\n')

telegram_specification = telegram_specifications.V4
telegram_specification['checksum_support'] = True

parser = TelegramParser(telegram_specification)
telegram = Telegram(sample, parser, telegram_specification)

print(telegram)
```

sander76, 2021-09-04:
I'm so sorry. I forgot to mention I found out myself what you just wrote. I was under the impression the "slimme meter" was able to determine both the amount of power being generated by the solar panels AND the amount of power delivered back to the grid. I was wrong.
Many thanks! As far as I see you can close this issue. My input data proves DSMR 4.2.2 IS supported.

closed 2022-10-07 by ndokter