ndokter/dsmr_parser #84: DSMR read problem

H35am, 2021-08-02:
DSMR wont read under Home Assistant and gives always the same error.

Logger: dsmr_parser.clients.protocol
Source: /usr/local/lib/python3.9/site-packages/dsmr_parser/clients/protocol.py:121
First occurred: 02:07:00 (1308 occurrences)
Last logged: 12:34:42

disconnected due to exception
Traceback (most recent call last):
File "/usr/local/lib/python3.9/site-packages/serial_asyncio/init.py", line 114, in _read_ready
data = self._serial.read(self._max_read_size)
File "/usr/local/lib/python3.9/site-packages/serial/serialposix.py", line 595, in read
raise SerialException(
serial.serialutil.SerialException: device reports readiness to read but returned no data (device disconnected or multiple access on port?)

This problem seems to be a while now with no clear solution:
https://community.home-assistant.io/t/dsmr-slimme-meter-problem/15956/4
https://community.home-assistant.io/t/dsmr-broken/255149

timovd, 2021-11-18:
Looks like I have the same problem when using a(n active) P1 splitter.

dupondje, 2023-11-09:
Seems like an issue with receiving the data.
Test with screen / minicom / whatever if you really receive telegrams.
Please reopen if still broken.

closed 2023-11-09 by dupondje