ndokter/dsmr_parser #95: Naming clashes in obis_name_mappings.py

bremme, 2021-11-22:
The `EN` dictionary in `obis_name_mappings.py` contains duplicated keys. The following keys are duplicated:

* ELECTRICITY_IMPORTED_TOTAL
* LUXEMBOURG_ELECTRICITY_USED_TARIFF_GLOBAL
* SWEDEN_ELECTRICITY_USED_TARIFF_GLOBAL

All have the same value: `r'\d-\d:1\.8\.0.+?\r\n'` .

* ELECTRICITY_EXPORTED_TOTAL
* LUXEMBOURG_ELECTRICITY_DELIVERED_TARIFF_GLOBAL
* SWEDEN_ELECTRICITY_DELIVERED_TARIFF_GLOBAL

All have the same value: `r'\d-\d:2\.8\.0.+?\r\n'`

It seems that these three different names indicate the same values? If that's the case I guess it makes sense to skip the Luxembourg and Sweden names from the `obis_name_mappings`. Let me know if I can be of any help, I'm willing to open a PR if that seems useful to you?

bremme, 2021-11-22:
Perhaps its also a good idea to check `obis_reference.py` for duplicated using a unit test to prevent this from happening in the future?

ndokter, 2021-11-22:
@lowdef also noticed this 2 days ago: https://github.com/ndokter/dsmr_parser/pull/92#discussion_r753721127

I missed it myself, but it seems something that would be nice to clean up. Only problem is that these values might now be in use, so we might want to make it backwards compatible? What do you think

bremme, 2021-11-22:
Perhaps a nice and simple solution is to just flip the mappings; use a dictionary with `obis_reference: name` and reverse that:

```python
from dsmr_parser import obis_references as obis

"""
dsmr_parser.obis_name_mapping
~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~

This module contains a mapping of obis references to names.
"""

REVERSE_EN = {
    'P1_MESSAGE_HEADER': obis.P1_MESSAGE_HEADER,
    'P1_MESSAGE_TIMESTAMP': obis.P1_MESSAGE_TIMESTAMP,
    'ELECTRICITY_IMPORTED_TOTAL': obis.ELECTRICITY_IMPORTED_TOTAL,
    'ELECTRICITY_USED_TARIFF_1': obis.ELECTRICITY_USED_TARIFF_1,
    'ELECTRICITY_USED_TARIFF_2': obis.ELECTRICITY_USED_TARIFF_2,
    'ELECTRICITY_EXPORTED_TOTAL': obis.ELECTRICITY_EXPORTED_TOTAL,
    'ELECTRICITY_DELIVERED_TARIFF_1': obis.ELECTRICITY_DELIVERED_TARIFF_1,
    'ELECTRICITY_DELIVERED_TARIFF_2': obis.ELECTRICITY_DELIVERED_TARIFF_2,
    'ELECTRICITY_ACTIVE_TARIFF': obis.ELECTRICITY_ACTIVE_TARIFF,
    'EQUIPMENT_IDENTIFIER': obis.EQUIPMENT_IDENTIFIER,
    'CURRENT_ELECTRICITY_USAGE': obis.CURRENT_ELECTRICITY_USAGE,
    'CURRENT_ELECTRICITY_DELIVERY': obis.CURRENT_ELECTRICITY_DELIVERY,
    'LONG_POWER_FAILURE_COUNT': obis.LONG_POWER_FAILURE_COUNT,
    'SHORT_POWER_FAILURE_COUNT': obis.SHORT_POWER_FAILURE_COUNT,
    'POWER_EVENT_FAILURE_LOG': obis.POWER_EVENT_FAILURE_LOG,
    'VOLTAGE_SAG_L1_COUNT': obis.VOLTAGE_SAG_L1_COUNT,
    'VOLTAGE_SAG_L2_COUNT': obis.VOLTAGE_SAG_L2_COUNT,
    'VOLTAGE_SAG_L3_COUNT': obis.VOLTAGE_SAG_L3_COUNT,
    'VOLTAGE_SWELL_L1_COUNT': obis.VOLTAGE_SWELL_L1_COUNT,
    'VOLTAGE_SWELL_L2_COUNT': obis.VOLTAGE_SWELL_L2_COUNT,
    'VOLTAGE_SWELL_L3_COUNT': obis.VOLTAGE_SWELL_L3_COUNT,
    'INSTANTANEOUS_VOLTAGE_L1': obis.INSTANTANEOUS_VOLTAGE_L1,
    'INSTANTANEOUS_VOLTAGE_L2': obis.INSTANTANEOUS_VOLTAGE_L2,
    'INSTANTANEOUS_VOLTAGE_L3': obis.INSTANTANEOUS_VOLTAGE_L3,
    'INSTANTANEOUS_CURRENT_L1': obis.INSTANTANEOUS_CURRENT_L1,
    'INSTANTANEOUS_CURRENT_L2': obis.INSTANTANEOUS_CURRENT_L2,
    'INSTANTANEOUS_CURRENT_L3': obis.INSTANTANEOUS_CURRENT_L3,
    'TEXT_MESSAGE_CODE': obis.TEXT_MESSAGE_CODE,
    'TEXT_MESSAGE': obis.TEXT_MESSAGE,
    'DEVICE_TYPE': obis.DEVICE_TYPE,
    'INSTANTANEOUS_ACTIVE_POWER_L1_POSITIVE': obis.INSTANTANEOUS_ACTIVE_POWER_L1_POSITIVE,
    'INSTANTANEOUS_ACTIVE_POWER_L2_POSITIVE': obis.INSTANTANEOUS_ACTIVE_POWER_L2_POSITIVE,
    'INSTANTANEOUS_ACTIVE_POWER_L3_POSITIVE': obis.INSTANTANEOUS_ACTIVE_POWER_L3_POSITIVE,
    'INSTANTANEOUS_ACTIVE_POWER_L1_NEGATIVE': obis.INSTANTANEOUS_ACTIVE_POWER_L1_NEGATIVE,
    'INSTANTANEOUS_ACTIVE_POWER_L2_NEGATIVE': obis.INSTANTANEOUS_ACTIVE_POWER_L2_NEGATIVE,
    'INSTANTANEOUS_ACTIVE_POWER_L3_NEGATIVE': obis.INSTANTANEOUS_ACTIVE_POWER_L3_NEGATIVE,
    'EQUIPMENT_IDENTIFIER_GAS': obis.EQUIPMENT_IDENTIFIER_GAS,
    'HOURLY_GAS_METER_READING': obis.HOURLY_GAS_METER_READING,
    'GAS_METER_READING': obis.GAS_METER_READING,
    'ACTUAL_TRESHOLD_ELECTRICITY': obis.ACTUAL_TRESHOLD_ELECTRICITY,
    'ACTUAL_SWITCH_POSITION': obis.ACTUAL_SWITCH_POSITION,
    'VALVE_POSITION_GAS': obis.VALVE_POSITION_GAS,
    'BELGIUM_HOURLY_GAS_METER_READING': obis.BELGIUM_HOURLY_GAS_METER_READING,
    'LUXEMBOURG_EQUIPMENT_IDENTIFIER': obis.LUXEMBOURG_EQUIPMENT_IDENTIFIER,
    'LUXEMBOURG_ELECTRICITY_USED_TARIFF_GLOBAL': obis.LUXEMBOURG_ELECTRICITY_USED_TARIFF_GLOBAL,
    'LUXEMBOURG_ELECTRICITY_DELIVERED_TARIFF_GLOBAL': obis.LUXEMBOURG_ELECTRICITY_DELIVERED_TARIFF_GLOBAL,
    'SWEDEN_ELECTRICITY_USED_TARIFF_GLOBAL': obis.SWEDEN_ELECTRICITY_USED_TARIFF_GLOBAL,
    'SWEDEN_ELECTRICITY_DELIVERED_TARIFF_GLOBAL': obis.SWEDEN_ELECTRICITY_DELIVERED_TARIFF_GLOBAL,
    'Q3D_EQUIPMENT_IDENTIFIER': obis.Q3D_EQUIPMENT_IDENTIFIER,
    'Q3D_EQUIPMENT_STATE': obis.Q3D_EQUIPMENT_STATE,
    'Q3D_EQUIPMENT_SERIALNUMBER': obis.Q3D_EQUIPMENT_SERIALNUMBER,
}

EN = dict([(v, k) for k, v in REVERSED_EN.items()])

```

On the other hand, this does not solve my problem, but perhaps there is another solution to this. I would like to get a list of names for a given protocol version?

I tried this:

```python
from dsmr_parser import obis_name_mapping, telegram_specifications

version = "V5"

telegram_specification = getattr(telegram_specifications, version)

obis_names = [
    obis_name_mapping.EN[k] for k in telegram_specification["objects"]
]
```

But then I ran into these clashed. Flipping it around does not solve my issue since the telegram specification is specified by obis reference and not by name.

bremme, 2021-11-22:
Another possible solution would be to make all the references in `obis_references` an `Enum`:

```python
class ObisReference(Enum):
    ELECTRICITY_IMPORTED_TOTAL = r'\d-\d:2\.8\.0.+?\r\n'
    LUXEMBOURG_ELECTRICITY_USED_TARIFF_GLOBAL = r'\d-\d:2\.8\.0.+?\r\n'
    SWEDEN_ELECTRICITY_USED_TARIFF_GLOBAL = r'\d-\d:2\.8\.0.+?\r\n'
```

By doing to the reference is unique again and you van use its `name` or `value` separate. On the other hand this is all mainly for backward compatibility. For a new major version it probably makes more sense to use the same name for the same obis reference code.

bremme, 2021-11-22:
As an alternative to this a plain class or dataclass should work:

```python
@dataclass
class ObisReference:
    name: str
    regex: str
```

lowdef, 2021-11-22:
Lots of ideas ;-)

My opinion is that we should only have 1 and only 1 occurence of a unique reference. (It is unique ID in a global standard for a reason.)
I would prefer to use the name from the global standard ISO/IEC 62056-61 unfortunately i do not have a copy of that. So I can not check.

I would be gratefull if you fix this :-) And you have my support.

bremme, 2021-11-22:
@lowdef I totally agree, if possible stick to the ISO/IEC 62056-61 standard. I did a little bit of research and its not easy to find a naming scheme which gives an nice single name to an OBIS object.

I did found however some interesting documentation:

* DLMS - Blue Book Ed. 13 - https://www.dlms.com/files/Blue_Book_Edition_13-Excerpt.pdf
   * I'm pretty sure this document is the base for the ISO/IEC 62056-61 standard
 * especially chapter 7: "COSEM Object Identification System (OBIS)" seems to be relevant
* DLMS - Interface classes - OBIS codes - Data types - https://www.dlms.com/eng/cosem-object-definitions-46679.php
  * You can download the "COSEM Object definitions" files, but I think you need some other `.dat` files and it looks messy on Libre Office on my machine.

In any case I get the feeling that there is no official "name" for an OBIS objects. But rather there is a somewhat official name for the individual components. 

If I understand it correctly the OBIS codes is build using groups like so: `A-B:C.D.E*F`. If I look at `CURRENT_ELECTRICITY_USAGE = r'\d-\d:1\.7\.0.+?\r\n'`, C=1, D=7 and E=0. If I look up these symbols in the "COSEM Object definition" I find this:

* C=1 (for A=1): "ΣLi Active power+ (QI+QIV)" (Table 50)
* D=7 (for A=1): "Instantaneous value" (Table 51)
* E=0 (for A=1): "Total (fundamental + all harmonics)" (Table 53)

From what I've found this is as close I could get to something "official". If this is the case, its also somewhat problematic since some sort of translation/convention has to be made.

A convention should be defined which can be used to build one or more names for the OBIS object. Perhaps you want a short name, long name, id (`A-B:C.D.E*F`) and regex.

This seems something that would be nice to improve upon. I'm only a little bit worried about backward compatibility, which parts of the library are considered part of the public interface? Are the obis_reference and obis_name_mapping considered public for example?

lowdef, 2021-11-22:
Here is a similar overview: https://onemeter.com/docs/device/obis/

The naming is a bit deviating of what we already have, I suggest we align with what we have otherwise we have to start renaming everything. That would really break a lot.

We have:
```python
ELECTRICITY_USED_TARIFF_1 = r'\d-\d:1\.8\.1.+?\r\n'
ELECTRICITY_USED_TARIFF_2 = r'\d-\d:1\.8\.2.+?\r\n'
ELECTRICITY_DELIVERED_TARIFF_1 = r'\d-\d:2\.8\.1.+?\r\n'
ELECTRICITY_DELIVERED_TARIFF_2 = r'\d-\d:2\.8\.2.+?\r\n'
```

It would be logical to use the same naming convention for the 2 global extensions:
```python
ELECTRICITY_USED_TOTAL = r'\d-\d:1\.8\.0.+?\r\n'
ELECTRICITY_DELIVERED_TOTAL = r'\d-\d:2\.8\.0.+?\r\n'
```
This way people will understand the code better, and the interface is consistent.

PR #96 (2021-11-28): resolve name clash as reported in issue #95
Note this change resolves the name clash as reported in #95 and closes #95.

For the Luxembourg und Sweden cases a backward incompatible update is necessary:
| Old Name | New Name|
| ------------- | ------------- |
| LUXEMBOURG_ELECTRICITY_USED_TARIFF_GLOBAL  | ELECTRICITY_IMPORTED_TOTAL |
| LUXEMBOURG_ELECTRICITY_DELIVERED_TARIFF_GLOBAL | ELECTRICITY_EXPORTED_TOTAL |
| SWEDEN_ELECTRICITY_USED_TARIFF_GLOBAL  | ELECTRICITY_IMPORTED_TOTAL |
| SWEDEN_ELECTRICITY_DELIVERED_TARIFF_GLOBAL | ELECTRICITY_EXPORTED_TOTAL |
closed 2021-11-28 by ndokter