ndokter/dsmr_parser #97: Support DSMR data read via RFXtrx with integrated P1 reader

rhpijnacker, 2021-12-30:
Recently I purchased a [RFXtrxLAN](http://www.rfxcom.com/epages/78165469.sf/nl_NL/?ObjectPath=/Shops/78165469/Products/21201) with both a RFXtrx443XL and a [DSMR P1 cable](http://www.rfxcom.com/epages/78165469.sf/nl_NL/?ObjectPath=/Shops/78165469/Products/19602) installed.

This device reads the telegram data from the P1 port and wraps it line-by-line into packets that fit the RF packet protocol that is normally pushed onto its USB or network connection. 
One such RF packet looks like this:
* A length byte (the size of the _rest_ of the RF packet)
* A packettype byte, this is 0x62 for "async transmit/receive", used also for the DSMR telegram lines
* A subtype byte (01 for "P1", 02 for Teleinfo, 03 for "raw" receive (not yet used)
* A sequence number byte (wraps when it gets to 255)
* The rest are data-bytes, in the case of a P1 packet a single line of telegram data.
All of this is properly documented in the RFXcom SDK documentation.

Over the last week(s) I've been trying to get this data into Home Assistant by feeding it through the RFXtrx integration. See e.g. https://github.com/Danielhiversen/pyRFXtrx/issues/125.
However, the RFXtrx integration expects one RF packet to equal a complete sensor update, including identifier. Fitting in the case where multiple "RF" packets form a single telegram that includes the serial identifiers is turning out to get complicated.

I considered starting a new Home Assistant integration that handles this particular combination of RFXtrx and DSMR and started looking at the DSMR integration to see how things are done there. I realised that it would be trivial if we could simply add a variation where the RF packets are interpreted, skipping RF packets that aren't DMSR data and collecting the telegram data for ones that are.

If I would implement this variation, would this make any chance of getting accepted into this library?

rhpijnacker, 2021-12-30:
Implemented a proof-of-concept, which works on my RFXtrx setup (with an additional single-line hack on the protocol level: changing DSMRProtocol into RFXtrxDSMRProtocol).
I can add the DSMR integration in Home Assistant without any problems after loading this change in my Home Assistant test environment, and the measured values are getting updated.

To finish the PR we would need to decide on what the external API should be:
* Add additional functions to the API (e.g. `create_rfxtrx_dsmr_reader` and `create_rfxtrx_tcp_dsmr_reader`)
* Add an additional parameter that specifies the whether or not to use the RFXtrx wrapper (could default to False)

ndokter, 2021-12-30:
Would it be possible to put the RFX specific part into a (new?) client module? It could inherit from the DSMRProtocol if needed i think. This way the RFX specific code is contained in it's own module and the default library is unaffected.

rhpijnacker, 2021-12-30:
Yes, I think that could be possible.
Do you have any suggestions for the file structure and API you'd like to see?
E.g. something like rfxtrx_protocol.py containing `create_rfxtrx_dsmr_reader`, `create_rfxtrx_tcp_dsmr_reader` and `create_rfxtrx_dsmr_protocol` would make sense to me.
This would probably give a small bit of code duplication that could be avoided by using an additional parameter.
Maybe we could keep the external API the same and create some private functions with that extra parameter to keep the duplication minimal?

ndokter, 2021-12-30:
Those names sound good! What part would need to be duplicated?

rhpijnacker, 2021-12-30:
In the current implementation only line 48 needs to be changed from DSMRProtocol into RFXtrxDSMRProtocol to make it work. Creating a private create_dsmr_protocol passing this as parameter would avoid duplication there.
The other two functions would need to call that with the appropriate parameter.

ndokter, 2021-12-30:
Maybe you can add a 'protocol' parameter?
`def create_dsmr_protocol(dsmr_version, telegram_callback, loop=None, protocol=DSMRProtocol , **kwargs):`

Which then is overrided only in the create_rfx.. function? Then the other methods won't work with the RFXtrxDSMRProtocol  object which might be confusing?

rhpijnacker, 2021-12-30:
I updated the (draft) Pull Request. Slightly differently than what you suggested. If you prefer, I can easily update it.

ndokter, 2021-12-30:
Nice! Same line of thought :)

rhpijnacker, 2021-12-30:
I tried following your suggestion, but somehow Python will not let me use `protocol=DSMRProtocol` in the function declaration. I do not know enough Python to understand why.

Is there anything I should change/add before you would accept the PR?

rhpijnacker, 2022-01-04:
Great to see the Pull Request getting merged. Thanks!
Do you plan to release this as a new version too?

ndokter, 2022-01-04:
Hey, yes i have to manually make a new release. Will do it soon. Thanks for the contribution!

closed 2022-01-04 by ndokter