ndokter/dsmr_parser #99: aurum warmtelink support (GJ)

pooh22, 2022-01-07:
I just got my warmtelink, it has a P1 port, but only gives GJ information from the "stadsverwarming"
```
/AUR2NWA-MYRSKY

1-3:0.2.8(50)
0-0:1.0.0(220107195809W)
0-0:96.1.1(0)
0-1:24.1.0(004)
0-1:96.1.0(123456789A1B234C)
0-1:24.2.1(220107195809W)(0.477*GJ)
!571
```
(the checksum doesn't work, I changed the serial number, but you get the point I assume)

It would be nice if dsmr_parser could understand this message and provide a usable interface for it. The device behaves like a normal DSMR 5 meter, but there's of course no electricity information.

I wouldn't be surprised if some setups combine this info via the DSMR electricity meter, just like with the gas meter info via the P4 port. 
So at some point, this type of info might show up in a "normal" DSMR reading as well.

lowdef, 2022-01-07:
Hi,

The parser can parse the telegram without any problem:
```
#!/usr/bin/env python3
# -*- coding: utf-8 -*-

from dsmr_parser import telegram_specifications
from dsmr_parser.objects import Telegram
from dsmr_parser.parsers import TelegramParser

TELEGRAM = r"""/AUR2NWA-MYRSKY

1-3:0.2.8(50)
0-0:1.0.0(220107195809W)
0-0:96.1.1(0)
0-1:24.1.0(004)
0-1:96.1.0(123456789A1B234C)
0-1:24.2.1(220107195809W)(0.477*GJ)
!571"""

sample = TELEGRAM.replace('\n', '\r\n')

telegram_specification = telegram_specifications.V5
telegram_specification['checksum_support'] = False

parser = TelegramParser(telegram_specification)
telegram = Telegram(sample, parser, telegram_specification)

print(telegram)
```

result:
```output
P1_MESSAGE_HEADER: 	 50	[None]
P1_MESSAGE_TIMESTAMP: 	 2022-01-07T19:58:09+01:00	[None]
EQUIPMENT_IDENTIFIER: 	 0	[None]
DEVICE_TYPE: 	 4	[None]
EQUIPMENT_IDENTIFIER_GAS: 	 123456789A1B234C	[None]
HOURLY_GAS_METER_READING: 	 0.477	[GJ] at 2022-01-07T19:58:09+01:00
```

The only thing is that the parser can not yet handle multiple channels. 
And it does not interpret the device type, which indicates the type of meter. 
To make that happen would be some major effort.
This is the reason that the reading is presented as a GAS_METER_READING, which is of course not correct.

pooh22, 2022-01-08:
Thanks for the explanation!

I guess I will just have to use the GAS_METER_READING, though I don't think it's hourly, but more like 5 minutes (also according to the DSMR document I have

> Last 5-minute Meter  0-n:24.2.1.255

It would be nice if the meter type is recognised at some point.
For my purposes, I can use the GJ entry and even calculate the m³ gas equivalent to satisfy home assistant....
I'll leave it up to you whether you want to leave this issue open or not.

lowdef, 2022-01-08:
Fixing this was something I thought about in the past.
As  I have no use for this myself and nobody seemed to have a problem i did not pursue it.
You're case shows that we need to do something here. 
For now please use the workaround you suggested.