ndokter/dsmr_parser #108: dsmr_parser.exceptions.ParseError: ("Invalid '%s' line for '%s'", '1-1:31.4.0(100.0*A,-100.0*A)\r\n',

kszys, 2022-06-12:
I have been using the DSMR Parser in Home Assistant for some time, with a Belgian electricity meter. Everything has been working fine until today. Today I get spammed by the errors:

```
Logger: dsmr_parser.parsers
Source: /usr/local/lib/python3.9/site-packages/dsmr_parser/parsers.py:61
First occurred: 15:27:52 (905 occurrences)
Last logged: 15:42:54

ignore line with signature \d-\d:31\.4\.0.+?\r\n, because parsing failed.
Traceback (most recent call last):
  File "/usr/local/lib/python3.9/site-packages/dsmr_parser/parsers.py", line 59, in parse
    telegram[signature] = parser.parse(match.group(0))
  File "/usr/local/lib/python3.9/site-packages/dsmr_parser/parsers.py", line 201, in parse
    return CosemObject(self._parse(line))
  File "/usr/local/lib/python3.9/site-packages/dsmr_parser/parsers.py", line 151, in _parse
    raise ParseError("Invalid '%s' line for '%s'", line, self)
dsmr_parser.exceptions.ParseError: ("Invalid '%s' line for '%s'", '1-1:31.4.0(100.0*A,-100.0*A)\r\n', <dsmr_parser.parsers.CosemParser object at 0xaf84ac58>)
```

Any ideas?

lowdef, 2022-06-12:
Well there is no pattern specified in the parser that knows how to parse this line '1-1:31.4.0(100.0*A,-100.0*A)\r\n'. Therefore you see this message, but this line is just ignored so it should not cause real problems.

Maybe the smartmeter has been upgraded and is now emitting this new line?

kszys, 2022-06-13:
Yes - thanks. I realised afterwards that essentially issue #72 was related to it. However, it is still not closed for some reason... Anyway - thanks. I thought the _other_ problems I was having was related to this error, but apparently not. 

BTW - if the line is ignored and it does not stop the parser from working and parsing other lines - perhaps this could be downgraded to a **Warning** instead of an **Error** - would not be as _scary_ when looking for issues in the logs ;)

dupondje, 2023-11-09:
Should be fixed since https://github.com/ndokter/dsmr_parser/commit/d05fe2692bce9da5bca4103008a6327b4901106a