ndokter/dsmr_parser #117: Support new OBIS 1.6.0 Belgium Flanders

Dr-know, 2022-12-29:
Hello,
As of 01/01/2023 the new OBIS code 1.6.0 is used to calculate the distribution tariff.
The OBIS code 1.6.0 (visible from firmware 199 ) is showing the current monthly calculated monthly peak usage in KW.
Every new month this calculated peak is reset to 0.
this is the "new" line i see in de debug of my Home assistant.
1-0:1.6.0(221203190000W)(04.035*kW)
So for the this month my maximum peak is 4.035 KW ( 4035 W)
Can this value be added to Home assistant ?
i can than create a helper to save each month the peak value, needed to calculate the distribution tariff .

If needed any additionally data pleas let me know !

dennissiemensma, 2022-12-29:
FYI: This was added already in #113 by @dupondje. 

For Home Assistant to support this you may want to create a feature request at the Home Assistant project instead. Bumping the `dsmr_parser` version used should suffice as a starting point for further integration in HA.

dupondje, 2022-12-30:
I am also preparing the HA side:
https://github.com/home-assistant/core/pull/84097

Dr-know, 2022-12-30:
Ok, nice to here that.
So I will Wait for ha to bump the dsmr_parser

closed 2022-12-30 by Dr-know