ndokter/dsmr_parser #127: "TypeError: 'str' object is not callable" in filereader.py

gsaviane, 2023-02-18:
I'm using FileReader to read telegrams from stdin. If a telegram has a checksum error it throws an exception, but the exception handling throws this:

> Traceback (most recent call last):
> 
>   File "/home/pi/dsmr2mqtt.py", line 101, in <module>
>     for telegram in telegrams:
>   File "/home/pi/.local/lib/python3.9/site-packages/dsmr_parser/clients/filereader.py", line 126, in read_as_object
>     logger.warning(str(e))
> TypeError: 'str' object is not callable
> 

I'm not a Pyhon expert, but it seems the str object gets overridden (lines 116-128 of filereader.py):

        with fileinput.input(mode='rb') as file_handle:
            while True:
                data = file_handle.readline()
                str = data.decode()
                self.telegram_buffer.append(str)

                for telegram in self.telegram_buffer.get_all():
                    try:
                        yield Telegram(telegram, self.telegram_parser, self.telegram_specification)
                    except InvalidChecksumError as e:
                        logger.warning(str(e))
                    except ParseError as e:
                        logger.error('Failed to parse telegram: %s', e)

ndokter, 2023-02-18:
Thank you for bringing this up. This however is caused by the same issue as #122

ndokter, 2023-02-20:
I'm sorry, i misread. This should indeed be fixed