ndokter/dsmr_parser #152: Add support for Iskra IE.5 (Belgian Smart Meter) - REW

ljehu, 2024-04-05:
I have a Iskra IE.5 meter installed.
Needed to modify the  obis_references.py file 

Line 106: 
From
BELGIUM_VERSION_INFORMATION = r'^\d-\d:96\.1\.4.+?\r\n'
BELGIUM_EQUIPMENT_IDENTIFIER = r'^\d-0:96\.1\.1.+?\r\n'
To
BELGIUM_VERSION_INFORMATION = r'^\d-\d:96\.14\.0.+?\r\n'
BELGIUM_EQUIPMENT_IDENTIFIER = r'^\d-0:96\.1\.0.+?\r\n'

ljehu, 2024-04-05:
Adding log with telegram sample:

2024-04-05 15:49:53.984 DEBUG (MainThread) [dsmr_parser.clients.protocol] got telegram: /ISk5\2MIE5E-200

1-0:0.0.0(00000000)
0-0:96.1.0(91318181)
0-0:1.0.0(240405154953)
1-0:1.8.1(000036.396*kWh)
1-0:1.8.2(000044.115*kWh)
1-0:2.8.1(000083.176*kWh)
1-0:2.8.2(000008.863*kWh)
0-0:96.14.0(0001)
1-0:1.7.0(00.000*kW)
1-0:2.7.0(01.568*kW)
1-0:21.7.0(00.000*kW)
1-0:22.7.0(01.568*kW)
1-0:32.7.0(236.8*V)
1-0:31.7.0(006*A)
0-0:96.3.10(1)
0-0:96.13.0()
0-1:96.1.1()
!6690

dupondje, 2024-04-05:
This is a new meter that gets installed in Belgium?

dupondje, 2024-04-05:
Ah, Wallonie :)
Feel free to create a PR!

ljehu, 2024-04-11:
Created PR #153.

closed 2024-04-11 by ljehu