ndokter/dsmr_parser #163: MicroPython support?

PolarGoose, 2024-07-15:
Is it possible to run this library, at least partially, on a MicroPython board like ESP32?

ndokter, 2024-07-15:
I have no idea. This is the simplest way to parse a stream of byte data (with the buffer helper), or if you skip that part, just put in complete telegrams (large strings) and it will get parsed. I don't believe the code in this example doesn't make use of advanced language features, so maybe it will work.

```
    def read_as_object(self):
        """
        Read complete DSMR telegram's from stdin of filearguments specified on teh command line
        and return a Telegram object.
        :rtype: generator
        """
        with fileinput.input(mode='rb') as file_handle:
            while True:
                data = file_handle.readline()
                str = data.decode()
                self.telegram_buffer.append(str)

                for telegram in self.telegram_buffer.get_all():
                    try:
                        yield self.telegram_parser.parse(telegram)
                    except InvalidChecksumError as e:
                        logger.warning(str(e))
                    except ParseError as e:
                        logger.error('Failed to parse telegram: %s', e)
```

PolarGoose, 2024-07-19:
@ndokter,

Thank you for your quick and detailed response.
I have tried to run your library using Micropython. Unfortunately, it seems it is not possible.
The Micropython doesn't have the following libraries: `copy`, `decimal`, `ctypes`
I haven't tried to check what else is missing.

ndokter, 2024-08-25:
I will close this for now as there are no plans currently to make it compatible

closed 2024-08-25 by ndokter